prusa3d/PrusaSlicer #9361: Please add paint feature for Skirt and Brim on individual objects

laptop-laptop, 2023-01-05:
**Is your feature request related to a problem? Please describe.**
I have a STL file with two objects. One I do not want to have skirt/brim but the other I want to add skirt/brim.
Currently, PS only allows either adding shirt/brim to all objects or none.

**Describe the solution you'd like**
Perhaps like Paint on Support, allow us to paint at the bottom of object we want to have skirt/brim but not the others.

sarusani, 2023-01-05:
You can already set a brim on individual objects.

Right click the object on which you want to add a brim. Select "Add settings -> "Skirt and brim". Tick "Brim width" -> OK. The set the brim with to your desired width.
The same way you can also change the brim type and separation gap per object.

mpaw, 2023-01-06:
Also if your two objects appear as one you can click on them and press the `O` button in a box at the top (next to the `P` which instead breaks it into individual parts for the same object).

[image]

laptop-laptop, 2023-01-06:
It is now working. Thank you very much.

closed 2023-01-06 by laptop-laptop