prusa3d/PrusaSlicer #9362: Faster slicing like SuperSlicer

laptop-laptop, 2023-01-05:
**Is your feature request related to a problem? Please describe.**
Given the same STL files, SuperSlicer always slices faster than PursaSlicer. 

**Describe the solution you'd like**
Could you please make PrusaSlicer slices faster like SuperSlicer? I think they started slicing even before the Slice button is pressed.

TojikCZ, 2023-01-05:
I do not know how SuperSlicer slices but there always has been a feature called "Background processing" just turn it on in Preferences under General. Let me know if that helps.

laptop-laptop, 2023-01-06:
Thank you. I timed the slicing process with and without the "Background processing" on. In both cases, it took PrusaSlicer 2.5 1:25 to complete the slicing. I then tried it on SuperSlicer 2.4.58.5 (Background processing off), it took only 00:11.54 to complete the same task. A very big different.

lukasmatena, 2023-01-06:
@laptop-laptop Please provide a 3mf. Thanks.

lukasmatena, 2023-01-06:
And by 3mf, I really mean one from PS and one from SS. The settings might be different, maybe you are comparing apples and oranges.

closed 2026-09-29 by prusaslicer-bot[bot]