prusa3d/PrusaSlicer #9363: CLI switch "--duplicate-grid" causes segmentation fault

el-bart, 2023-01-06:
### Description of the bug

adding `--duplicate-grid` (even `1,1`) causes segmentation fault:
```
$ prusa-slicer --loglevel 5 --export-gcode --output cube10.gcode --duplicate-grid 2,2 cube10.stl 
[2023-01-06 08:23:58.864508] [0x00007f0f52caf900] [trace]   Initializing StaticPrintConfigs
[2023-01-06 08:23:58.875792] [0x00007f0f52caf900] [debug]   TriangleMesh::repair() started
[2023-01-06 08:23:58.876418] [0x00007f0f52caf900] [debug]   TriangleMesh::repair() finished
Segmentation fault (core dumped)
```

using `--duplicate` works as expected:
```
$ prusa-slicer --loglevel 5 --export-gcode --output cube10.gcode --duplicate 4 cube10.stl
[2023-01-06 08:24:38.753564] [0x00007f698cf1f900] [trace]   Initializing StaticPrintConfigs
[...]
Slicing result exported to cube10.gcode
```

this is problematic, as `--duplicate` cannot be used with `--complete-objects` on tall models. however `--duplicate 1,4 --complete-objects`  would cut it.

### Project file & How to reproduce

example object: [cube10.stl.gz](https://github.com/prusa3d/PrusaSlicer/files/10358710/cube10.stl.gz)
though the problem seems to occur regardless of what model is used.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

linux, debian 11

### Printer model

anycubic chiron

isotropo, 2023-08-03:
Reproduced this error on MacOS Monterey. I am not super experienced with C++ but willing to try to investigate a little together if you are interested. I am located in LA so you're +9 hr? I have Ubuntu installed on a computer so we could both work from Linux if that makes things easier.

el-bart, 2023-08-10:
thx @isotropo for reaching out. i'm UTC+2h atm, however TBH i'm a bit short on time here with other projects, so that might be a bit difficult atm. :/

however if u happen to have SDK already setup, then just running debug-compiled code through `gdb` should be enough to capture backtrace of  the crash. if u could provide that, there's a good chance that having stacktrace alone could allow to spot a bug in the code. even better if it would be possible to compile with ASAN (address-sanitized) - no `gdb` needed there.

closed 2026-09-29 by prusaslicer-bot[bot]