prusa3d/PrusaSlicer #9365: Send System Info pop-up

jenwilsonglia, 2023-01-06:
### Description of the bug

When I click on "send system info" on this pop-up window, I'm then locked and can't get the pop-up down, or shut slicer down. If I click on "ask me next time" then it's fine, I can close the pop-up and work on slicer no problem. But after clicking "send system info", I have to reboot my laptop in order to close the pop-up (and slicer). 
[image]

### Project file & How to reproduce

1. Open Prusa Slicer
2. Click on "send system info" on the Send System Info pop-up screen
3. nothing will close pop-up now
4. need to reboot to even close slicer

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0Linux

### Operating system

Ubuntu 22.04.1 LTS

### Printer model

Prusa MK3S

Jan-Soustruznik, 2023-01-09:
hello, @jenwilsonglia, 

Thank you for report of issue. We will fix in next release.

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena