prusa3d/PrusaSlicer #9366: Missing profiles

Digitalpeer1, 2023-01-06:
### Description of the bug

I just recover the data from my computer after a hard reset and when I import the Prusa slicer .ini files the filament and print settings don't work on CUSTOM presets for a printer only on premade ones even when I change one thing about a premade one the filament and print settings don't show up the printer presets DO work but not the others

### Project file & How to reproduce

unknown
[image]
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11

### Printer model

modified ender3v2

kubispe1, 2023-01-09:
Hi, can you provide us more info. Could be also useful to provide us ini file.

lukasmatena, 2024-02-19:
No response, closing.

closed 2024-02-19 by lukasmatena