prusa3d/PrusaSlicer #9370: 2.5.0 error on startup I receive "Unhandled unknown exception; terminating the application"

MiFixIt87, 2023-01-07:
### Description of the bug

At startup I receive the following error
[image]

when I click OK to that I receive the following:

[image]

I have searched for information on this error and one thread indicated that it was related to an NVIDIA graphics setting.  I checked my NVIDIA settings and mine are correct.

[image]

My system is:

Prusa Slicer 2.5.0+Win 
Dell Precision 7760 
Intel UHD Graphics (only using the built-in display)
NVIDIA RTX A6000 graphics (No external displays connected)
32GB RAM
1TB NVMe SSD

I had this same issue with Prusa Slicer 2.4.0 and upgraded to 2.5.0 hoping it would cure the issue.
It never proceeds to a point where I can export the configuration or configure printers.

Any ideas?

### Project file & How to reproduce

Start Prusa Slicer or try to 3d print from Fusion 360 with Prusa Slicer set as the application to print to.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 Windows

### Operating system

Windows 11 pro

### Printer model

never gets that far before crashing

kubispe1, 2023-01-08:
Hi, there are quite similar issue:

MiFixIt87, 2023-01-08:
I saw that before I posted and I tried the solution that worked for the OP ( see the NVIDIA setting screenshot in my post) and it did not resolve my issue.

MiFixIt87, 2023-01-08:
Closed accidentally, issue is not resolved.

jrwaters2, 2023-01-12:
I'm having the same issue and my Nvidia settings are to allow the application to decide.

I did have an unexpected Windows shutdown earlier that I suspect is the trigger.

My ini file does not look corrupted - I didn't try to manually compare the MD5 but there is no garbage in the ini file.  Also the .ini.bak file matches the main .ini file.

jrwaters2, 2023-01-15:
The normal PrusaSlicer.exe also accepts the parameter --sw-renderer.  For now, I've just created a shortcut and all is working fine.  Of course would love to get back to normal rendering at some point, but for the desperate:

[image]

MiFixIt87, 2023-01-15:
I tried your workaround for my issue, and it works on my system as well.
If you do find a solution that allows for hardware graphics, please post it here or DM me if possible.
Thank you very much!!!!!

Tux0Topo, 2023-01-17:
Also having the same issue.
Notebook, with Windows 11, intel i7 11th gen + RTX 3080.
--sw-renderer is a bypass.
Any one knows if the lack of HW renderer will be that bad?

jrwaters2, 2023-01-18:
> Also having the same issue. Notebook, with Windows 11, intel i7 11th gen + RTX 3080. --sw-renderer is a bypass. Any one knows if the lack of HW renderer will be that bad?

Its been fine for me.  Not saying I like the idea but performance-wise I don't notice a difference for what I'm doing.

kubispe1, 2023-02-03:
Have you tried also 2.6.0? alfa 2? Please? Can you check ?

Tux0Topo, 2023-02-03:
Same error with Versión 2.6.0-alpha2+win64.

Tux0Topo, 2023-02-03:
alpha3 also having the same problem!

jrwaters2, 2023-02-04:
Of course - will definitely let you know.  Been experimenting with driver
upgrades.  So far no luck.

Jeroen6, 2023-02-06:
Can confirm this issue on Dell 3571, Windows 11 Pro (10.0.22621 Build 22621) and NVIDIA T600 driver 517.66 for prusaslicer 2.5.0 and alpha3.

Dell pushed driver updates for video in january. Maybe those are of help.
`NVIDIA Txx0/Txx00/RTX Ax00/RTX Ax000/GeForce MXx50 Graphics Driver - A04`
`Intel UHD/Iris Xe Graphics Driver and Intel Graphics Command Center Application - A09`

--sw-rendered does help

bubnikv, 2023-02-07:
Hopefully fixed with 84722876012ca310e29b291e10fef9d18ae26cea
Implementing a test whether the current thread is the main (UI) thread
and using it on AppConfig::save() to assert if save is called
from a worker thread.
The old assert was using thread names, which did not always work
on Windows.

MichalBuzek, 2023-02-10:
I have the same error with PrusaSlicer-2.6.0-alpha3+win64-202302031521_signed 
on 
Dell Precision 5570
 Windows 11 10.0.22621 N/A Build 22621
 Nvidia RTX A1000    Driver version: 31.0.15.1766
12th gen Intel i7-12800H
NVMe PC801
32GB ram

lukasmatena, 2023-02-10:
@MichalBuzek That is expected. @bubnik's fix was done after alpha3, so the first version to test the fix will be 2.6.0-alpha4.

Renbrant, 2023-02-14:
> --sw-renderer

This resource did not work for me.

Tux0Topo, 2023-02-20:
ITS FIXED ON ALPHA 4!!!
Great!!! Thanks!!