prusa3d/PrusaSlicer #9372: Feature reqeust: Add STEP support to Shape Gallery

Sembazuru, 2023-01-07:
**Is your feature request related to a problem? Please describe.**
Ever since PrusaSlicer started importing STEP files I've used that format between my CAD program (FreeCAD in my case) and PrusaSlicer. I recently created a shape that sliced and printed fine when loading the STEP file into PrusaSlicer. I want to put this shape into my Shape Gallery, but the add to gallery load window wouldn't show step files (only stl and obj). I tried to force the issue by copying the step file directly into the settings "shapes" folder, but no go. Looking at another stl file that I had imported into the shapes gallery I saw that a png thubnail had been created, so I created my own 256x256 resolution thumbnail in Gimp but the new shape still didn't show in the Shape Gallery. So I tried to fool the viewer and changed the extension of the step file from .step to .stl. Still no joy, it doesn't show in the Shape Gallery. (Note I did all the changes at the filesystem level with PrusaSlicer closed.)

**Describe the solution you'd like**
Add step support to the Shape Gallery, either by converting to stl and storing that in the configuration "shapes" folder, or store the base step file in the configuration "shapes" folder and tessellate when loading the shape. If storing the raw step folder, offer a scaling window when loading the object from the Shape Gallery to help avoid the issue noted in https://github.com/prusa3d/PrusaSlicer/issues/9184

**Describe how it would work**
When adding an object to the Shapes Gallery, step format is loaded just like stl and obj currently is.

**Describe alternatives you've considered**
For now I'll convert to mesh in FreeCAD and store the stl in the Shapes Gallery.

**Additional context**
Attached is the specific step object I'm trying to add to the Shapes Gallery (zipped because you haven't added STEP files to what this repository accepts for file attachments) just to make sure it isn't a weird STEP formatting issue from FreeCAD even though PrusaSlicer does load this file without issue:
[ForIndicationOnly-Extrude.zip](https://github.com/prusa3d/PrusaSlicer/files/10367236/ForIndicationOnly-Extrude.zip)

kubispe1, 2023-01-08:
Hi, .slt and .obj formats are supported for Gallery, as you observed. Thanks for the feature request

Sembazuru, 2026-09-03:
As a reminder, when getting around to implementing the Shape Gallery in PrusaSlicer 3.0.0, please make sure it can support any filetype that PrusaSlicer can import as geometry (or modifiers).

Jan-Soustruznik, 2026-09-08:
Hello, @Sembazuru , thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread, as I'd be interested to hear what the community thinks about how it should work. Thank you, and have a nice day.