prusa3d/PrusaSlicer #9373: Image on printed surfaces

rkarlsba, 2023-01-08:
Josef's twitter post https://twitter.com/Prusa3D/status/1611027418562826240 says it'll be possible to put text on curved surfaces. I couldn't find the bug here, so I'll just make a new one. Would it be hard to extend this to imported 2d images, say svg or png?

kubispe1, 2023-01-09:
There are some ideas around it. But for sure it will not be in the next release. Thanks for request
There is quite simillar issue: #4208

jkavalik, 2023-01-09:
Imho the linked issue is not very similar :)
When the embossing tool can do fonts, it can probably already handle many vectors (ofc depends on the actual implementation but I do not expect it to rasterize vector fonts), You could even try custom image fonts with the current version.
But getting full svg or even heightmap support would rock :)

gudvinr, 2023-01-10:
There is nanosvg library in dependencies. I believe it is used to load buildplate images.  
It is not actively maintained though and doesn't support text tags.

rkarlsba, 2023-01-10:
Interesting - the github page on https://github.com/memononen/nanosvg says it's not actively maintained, but the last change was a month ago. Dunno why tinysvg was chosen, though. libsvg is open
(also interesting, tinysvg has 301 forks!)

Jony01, 2023-02-02:
To use imge in svg One could create Font for this purpose: for example with [fontforge](http://fontforge.org/)

rcmaniac25, 2023-07-18:
Would much like this feature. Recently hit a moment where I needed the USB trident logo for a model. I could take the SVG, load it into Fusion 360 or something, give it depth, export the STL, import it into PrusaSlicer and cut/merge/etc. into the model.

Or just import the SVG and apply it like text. There's already SVG libraries built in, there's already vector -> mesh code built in. All that's needed is the ability to import an SVG itself and to connect the dots.

rkarlsba, 2023-10-11:
Any news on getting this into the slicer?

gudvinr, 2023-10-12:
> Any news on getting this into the slicer?

@rkarlsba, just subscribe and move on. When it's ready you'll know it from release notes and issue will be closed.  
Messages like this (including mine) just annoy everyone who already subscribed and bring nothing. It doesn't matter if it's going to be ready soon or not. You won't get it until it's done anyway.  
If you want it to be ready sooner, add it yourself and submit PR.

lukasmatena, 2023-10-21:
Implemented in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-10-21 by lukasmatena

nfd9001, 2023-10-22:
Looks pretty promising in change description in the slicer preview! A feature like this could be really great for a project I'm working on with an artist right now; sadly it isn't quite working with the feature that's out in this first slicer alpha. He's been using a drawing app on an Android tablet to draw baseplates and lineart, exporting rasters, and then using https://vectorizer.ai/ (any old tool like this could work) to export vectors I can work with more easily in e.g. Fusion360. Here's an actual example of a file I tried to import and what I got:

[image]
[image]

Not quite right.

This process obviously wouldn't create super-clean vectors with strictly only shape data, like one might get out of InkScape, so I wasn't expecting that it'd necessarily immediately work. I just think it'd be worth documenting what sorts of features are supported, and if being a bit more permissive of input in some specific way seemed worth a new issue, I could see about making one.

lukasmatena, 2023-10-22:
@nfd9001 Could you please open a separate issue and enclose the svg file so we can check it? Thanks.