prusa3d/PrusaSlicer #9377: Individual Temperatures For Multiple Filaments

JosephZicaro, 2023-01-09:
**Is your feature request related to a problem? Please describe.**
Yes. PrusaSlicer can't fully handle multi material units since it can't actually handle multi materials, e.g. PLA, PLA+, PETG, and PVA, at the same time because the slicer sets the same temperature for all filament/extruders, thus MMU is actually a poor name for such a device, at least for PrusaSlicer. Cura solves this problem but doesn't have the great painting features for multiple materials as PrusaSlicer does. 

**Describe the solution you'd like**
Please add an option to assign individual temperatures to either individual filaments or individual extruders. 

**Describe how it would work**
This would go under either "Filament" in "Filament Settings" or individual extruders, "Extruder 1," "Extruder 2,"... "Extruder n," in "Printer Settings". Under "Filament" seems like the best option, in which case this should probably be fit under the "Temperature" section with a drop-down menu at the top with two options, "Single Filament" and "Multiple Filaments", the former keeping all extra, new settings grayed out and the latter enabling the following new setting: numerical boxes appear for each of the filaments 1 through n, all with the original "Other layers" temperature setting autofilled in and editable. The first version of this setting should have a common "First layer" temperature setting for all filaments in the first layer, but a later version of this feature may have "First layer" and "Other layers" settings for each filament. Also note that it's probably best to reorder the "Bed" settings first and then put "Nozzle" setting, as well as all the above described new settings, below the drop-down menu, and all these new settings given the "Advanced" color code. 

**Describe alternatives you've considered**
From my perspective, there is no alternative. I'm only familiar with PrusaSlicer and Cura, and both do a good partial job of multi material printing, but neither does a good complete job. I'm whining about it here because PrusaSlicer seems much closer to ideal functionality than Cura. 

**Additional context**
-

rtyr, 2023-01-09:
>  the slicer sets the same temperature for all filament/extruders,

I don't understand,  that is not how it works. We are using temperatures from filament settings for each extruder individually.

[image]

JosephZicaro, 2023-01-09:
My mistake. Thanks for the picture. I just have to find out how to define temperatures for those filaments.

closed 2023-01-09 by JosephZicaro