prusa3d/PrusaSlicer #9378: Weird error message when shutting down PC

Plyn1x, 2023-01-09:
### Description of the bug

When prusaslicer is open and I shutdown my pc I get this notification 
I have an r5 3600
And an rx5700xt rocking 16gb ram with 3200mhz
[image]

### Project file & How to reproduce

Open prusaslicer load object then slice, the. Shut down 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Windows 10 latest update 

### Printer model

Every moddel  will work

bubnikv, 2023-01-10:
Please re-attach the error message screenshot.

Plyn1x, 2023-01-10:
[image]

Plyn1x, 2023-01-10:
[image]

mdvorak, 2023-01-19:
I've seen this plenty of times as well.

closed 2026-09-29 by prusaslicer-bot[bot]