prusa3d/PrusaSlicer #9381: Unable to drag and drop stls into slicer

halkony, 2023-01-10:
### Description of the bug

What is the problem?

I can't drag and drop an stl on Prusa Slicer 2.5.0. I see a sign showing me that it is disabled. My filesystem assosciates stls with Prusa slicer. I also have BambuStudio installed. I've had this problem since changing my .3mf association to Bambu.

[image]
[image]

### Project file & How to reproduce

[empty.zip](https://github.com/prusa3d/PrusaSlicer/files/10378776/empty.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version2.5.0-win64

### Operating system

Windows 10

### Printer model

prusa mk3s

kubispe1, 2023-01-11:
> I also have BambuStudio installed. I've had this problem since changing my .3mf association to Bambu.
So before this, you were able to drag and drop the stls that you are showing in the picture?
Send or attach your stls also and I can try, that's the only solution that comes to my mind.

halkony, 2023-01-12:
At one point I was able to drag and drop, yes. Then I started working with Bambu Studio and it stopped. I have tried every combination of file assosciations between .3mf and .stl and still it does not work.

It is true of every file on my system so sending an stl wouldn't be very helpful.

halkony, 2023-01-12:
Downloading the newest release from github and running it portably fixed the issue. Very confused.

kubispe1, 2023-01-19:
Dont have idea how to reproduce it. But glad to hear that you fix it.

libelred, 2023-01-21:
same issue here, it was working, a couple of days ago i had the issue, but resolved itself after restarting the slicer, but now is not working even after restarting the computer. gonna check with the latest release

libelred, 2023-01-21:
so i overwrote the original installation folder, and it didn't initially work. i noticed that when it launched after the extract, it prompted for admin rights (my guess is that this is related somehow) drag and drop didnt work.
i extracted the latest release to a different location and relauched the original location and drag and drop worked fine, this time didnt prompt for admin rights
i have closed and relauch several times, and is still working

libelred, 2023-01-21:
ok i just realized something else, i was trying to drag and drop an stl, that was inside a zip file, but the zip file was opening inside the windows explorer, so maybe this is the actual issue and is more a Windows issue, not prusa slicer

halkony, 2023-01-21:
> ok i just realized something else, i was trying to drag and drop an stl, that was inside a zip file, but the zip file was opening inside the windows explorer (also the zi had subfolders, and the stl was inside one of them), so maybe this is the actual issue and is more a Windows issue, not prusa slicer

I assumed you were correct about it being a Windows issue, but I just tested an .stl drag and drop directly from 7zip/WinRAR into PrusaSlicer and it worked. You should be able to do that too.

libelred, 2023-01-21:
> I assumed you were correct about it being a Windows issue, but I just tested an .stl drag and drop directly from 7zip/WinRAR into PrusaSlicer and it worked. You should be able to do that too.

yes that works

kubispe1, 2023-01-22:
Thanks guys to ensure and solve your problem. Seems to me that I can close the issue...

closed 2023-01-22 by kubispe1

SAM4191, 2023-07-16:
I have the same issue and realized that drag and drop doesn't work when prusaslicer is started as admin but is perfectly fine without admin rights.