prusa3d/PrusaSlicer #9382: ̶П̶о̶ч̶е̶м̶у̶  в слайсерах отсутствует резка крупногабарита из коробки:

Chernysh-Aleksandr-Nikolaevich, 2023-01-11:
̶П̶о̶ч̶е̶м̶у̶  в слайсерах отсутствует резка крупногабарита из коробки:
намечаешь плоскость, выбираешь размер кривой, тип замков (пазл, ласточкин хвост, лего и тд.), размерность (1d - один рез(на две детали), 2d - рез по принципу пазла, 3d - рез по принципу лего), тип фиксации (клеевой, под штифты, под метизы)
печать реза осуществляется как внешний контур (повышается прочность соединения - так как вдоль всего слоя соединения одна линия)
экономия времени монотонной работы в сапр если модель не влазит в слайсер (с компенсацией усадки)
возможность вращения отдельного блока модели (для чистовых поверхностей - если до этого не додумаются)
возможность смены материалов для разных блоков
возможность раздельного сохранения нарезанных блоков (+ версии слайсера и параметров нарезки)
и тд.

kubispe1, 2023-01-11:
Even if it is difficult to translate, text and ideas are structured strangely. Can you translate it into English and enhance your ideas with some pictures, please? 
Thanks

Chernysh-Aleksandr-Nikolaevich, 2023-01-11:
[image]

SphaeroX, 2023-01-12:
I think what he means is that I can split objects that are larger than the print bed into several parts with one click. 
In addition, the separated layer should have a toothing so that you can put them together again afterwards. Funnily enough, I just had the same idea, which would be perfect if you have large objects. At the moment I do it all myself in the CAD program. 

It would be important to first align the object, generate support if necessary, and then cut it.

SphaeroX, 2023-01-12:
From This
[image]

To This
[image]
[image]

JohnOCFII, 2023-01-12:
I have to admit -- that'd be pretty awesome.  Even to have that capability as a separate tool would be great -- the inclusion of joinery is key.

SphaeroX, 2023-01-29:
https://www.tomshardware.com/news/new-features-added-to-prusaslicer-26-alpha-release

Looks like it coming to 2.6 🧐

cybercop23, 2023-05-18:
Using 2.6 beta1 has 2 options for making connectors. I'm looking at the same idea of dove-tails. I'm trying to print a large thin object and dowels won't work in my case. Since (in my case) the object is flat and slim, Adding a cut with dovetails would be awesome.
Here's an example of what the cut may look like.
[image]

mfraser, 2023-09-03:
PrusaSlicer 2.6.1 now has dovetail joints. Is it possible to add more than 1 dovetail joint to a cut for larger pieces? Or is there another way of cutting a large piece into quarters and have a dovetail joint on each side?

adenta, 2024-01-18:
@mfraser  did you find a solution to multiple dovetail joints?

mfraser, 2024-01-18:
> @mfraser  did you find a solution to multiple dovetail joints?

No, haven't tried for a while.

closed 2026-09-29 by prusaslicer-bot[bot]