prusa3d/PrusaSlicer #9383: Peal Force Estimation Display

ShadowLordAlpha, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
It is a bit annoying somewhat guessing about supports and while I am getting good at it along with the auto supports not being bad it would be much nicer to see estimated numbers in the slicer for the forces that are happening

**Describe the solution you'd like**
Display an estimated peal force to the user for a layer of the print. Most likely based off of the type of mechanism used (tilt or lift) and how much of the build plate is covered and speed of the movement as well as an estimate for how much peal force the supports can take at a layer

**Describe how it would work**
An estimation of the peal force that occurs when a resin printer lifts or tilts the print off the plate and the support force

**Describe alternatives you've considered**
Rotating the object and looking at how much of an object is currently on the flat face while guessing with knoledge gained over time

**Additional context**
This would probably be best as a more Expert field and maybe added as an option to minimize peal forces in auto rotation or to allow the user to adjust supports to have more than needed based off the peal force of an object

murk-sy, 2023-01-15:
I've actually been thinking about this a bit and although I'm not sure how Prusaslicer automated supports work, there's guaranteed a way to get more reliable results, especially when you have things like 10° overhangs which are quickly ignored by automated support generation.

The simplest way is just counting the pixels and just calculating the max average peel force based on that. Basically, every layer below (or rather, before) a layer with the maximum pixel count *must* be able to support the peel force of that layer.

The "default" peel force of each pixel is 1. This means that a cube would make consistent peel force of 1 on each pixel of the model. However, an hourglass with widest area of 1000 pixels and narrowest area of 50 pixels would have the highest pull force of 20.

[image]

However, there's an obvious problem in this example already. You don't need to be an expert to see the A point (which may as well be multiple supports) has to support (or rather, resist the pull) more than each individual B point.

Ideally, the (probably?) best way to go about it is to calculate the maximum force by just going through pixels from the very end to the start of the print.

[image]

Hopefully this illustrates the point well. There are, of course, a few parameters that are more or less ignored in this, such as the angle at which the pull force propagates into the base of the model, kind of like angle of repose I suppose. This was done very roughly by hand to illustrate my point, but it should be obvious as there are some pixels with pull force of 1 right next to ones with a much higher force.

To be clear: this does not prevent unsupported parts from sagging, but it would help prevent delamination.

ShadowLordAlpha, 2023-01-15:
Interesting. While my idea had more to do with just showing the overall data to the user yours seems to have more to do with improving the support algorithms.

If thats the case have you tried doing the oposite calculation? instead of determining the amount of peal force per layer determin how much it is able to support? That way after processing each layer you not only know where the max support is needed by but also it should easily be able to detect all islands as well as where you need to add support structures and how much is needed (+ extra for safety)

murk-sy, 2023-01-15:
Ideally yes, the algorithm would be used in reverse to tune support structures - but the biggest disadvantage in its current (theoretical) form is that it requires slicing before supports are added or potentially made. It would work as a proofing step, but it wouldn't be the best user experience at least.

One thing to keep in mind is that this is far from a proper physics simulation, but it may be a good and resource cheap way to achieve similar results.

ShadowLordAlpha, 2023-01-16:
not really, should be easy enough to have it run along with the slicing. at most you would notice a little longer slicing time

closed 2026-09-29 by prusaslicer-bot[bot]