prusa3d/PrusaSlicer #9385: Arachne engine with spiral vase; extrusion_width is used after bottom_solid_layers

Oxmaster, 2023-01-11:
### Description of the bug

I had tolerance problems with spiral vase when mating 2 different prints.

I noticed that no matter the exported file wall size (included 0.45, 0.5, 0.6 and 0.7 wall thickness) the amount of filament used doesn't change much. The model is just a straight cylinder.
[image]

I thought that the new arachne engine would change the extrusion width to match model size, but it only happens for non spiral part of the print (after `bottom_solid_layers` it changes to `extrusion_width`).
Sliced info of 4 different models with sipral vase mode.
[image]

### Project file & How to reproduce

[prusaslicer-arachne-spiral-vase.zip](https://github.com/prusa3d/PrusaSlicer/files/10395925/prusaslicer-arachne-spiral-vase.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 11

### Printer model

Ender 3X super modded

kubispe1, 2023-01-17:
Hope I understand it clearly. So you are trying to push PS to make single spiral wall tickness by stl model wall tickness. 
I want you refer you to our help https://help.prusa3d.com/article/layers-and-perimeters_1748#spiral-vase. There is this note to spriral mode: 

> The model should be defined as a solid. Otherwise, PrusaSlicer will try (and fail) to create both inside and outside surfaces, so model the outside dimensions only.

So this is recommended approach:

- With solid (fullfill) model  you switch to spiral mode
- then you control width of first layer and width of External perimeters
- Also you can decrease count of bottom horizontal shells....

tg73, 2024-02-02:
PS no longer fails if you have a non-solid model - it seems to intelligently use only the outermost perimeter. It will probably still fail if there is more than one "island" in any spiralized layer.

I also would like to be able to control wall thickness according to the model wall thickness - so that wall thickness can be varied arbitrarily throughout the model. The simple approach of setting the External Perimeter width in PS does not achieve this.

closed 2026-09-29 by prusaslicer-bot[bot]