prusa3d/PrusaSlicer #9386: FLSUN V400

V4fun, 2023-01-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

V4fun, 2023-01-12:
Will the FLsun V400 profile be added to Slicer soon? I'm close, still having a couple issue getting the printer to print correctly every time.

rtyr, 2023-01-12:
We currently don't have any  pull requests for that printer (3rd party profiles are usually added based on community contributions).

V4fun, 2023-01-12:
Well, here it is if you ever need it.

Vince

rtyr, 2023-01-12:
You have to attach the files via web interface. E-mail attachments are unfortunately not supported.

lukasmatena, 2023-02-06:
Closing as stale.

@V4fun You can still send the profiles in here if you decide to, or send them as a pull request. Thanks.

closed 2023-02-06 by lukasmatena