prusa3d/PrusaSlicer #9388: z-dithering

LRaiz, 2023-01-12:
**Z- dithering**

The algorithm of z-dithering is proposed for incorporation into PrusaSlicer. As far as I am aware, no one implemented anything similar, and implementing it would be advantageous for PrusaSlicer. The z-dithering may be used in the case of filament deposition printing. It reduces the stairstep effect of layers near the local minimum and maximum of a printed surface. If printing is done with a layer height parameter, z-dithering produces stairsteps that are only half as tall. Only an area close to a volume surface is affected by the algorithm; the rest of the layer is printed with full layer thickness. Thus the approach allows improvement of resulting surface quality without a significant increase in print time, alternatively, considerable reduction of print time without much decrease of surface quality by using thicker layers. 

[image]

The illustration above assumes that regular (non-z-dither) printing computes slices in the middle of each layer. Then, for z-dithering additional slices are calculated between layers. Subsequently, the boolean operations are used on computed cuts to figure out the five regions indicated in the above picture:

1. Deposited on top of the previous layer, 0.25 layer height. Computed by subtracting the cut at the layer top from the layer bottom cut.
2. Deposited on top of region 1, 0.5 layer height. Computed by subtracting the cut at the layer top from the layer middle cut. 
3. Deposited on the top of a layer below, 1.0 layer height. Computed as an intersection of the cut above and cut below the layer.
4. Extruded from 0.75 layer height above layer bottom, 0.5 layer height. Computed by subtracting the cut at the bottom from the cut in the middle.  
5. Extruded from the top of a layer, 0.25 layer height. Computed by subtracting the middle from the top cut.

It is possible to have cuts at a top/bottom of a layer to coincide with a middle cut along part of the profile but be quite different along other portions. Then z-dithering will affect only areas of difference. Since the computation of regions 1-5 involves boolean operations between polygons that may partially overlap, the implementation of z-dithering depends on the reliability and robustness of regularized boolean operations on polygons.

It seems clear that z-dithering is useful on surfaces facing up; however, its usefulness on bottom-facing surfaces is questionable unless support is also printed with z-dithering or bridging prevents regions 4-5 from falling.

I assume that PrusaSlicer prints a single pass perimeter when an area to be covered is just a little wider than half of the nozzle diameter. For example, in the case of nozzle_diameter = 0.4mm and layer_height = 0.25mm, z-dithering starts making a difference by introducing region 1 when a surface becomes closer to horizontal than 50 degrees;  both regions 1 and 2 will be extruded for surfaces deviating from horizontal by less than 30 degrees. For side surfaces closer to vertical, z-dithering should not be applied because there is not enough space to fit even a single extruder pass into regions 2 and 4.

I pushed the basis of implementation to [https://github.com/LRaiz/PrusaSlicer.git]( https://github.com/LRaiz/PrusaSlicer.git) fork on GitHub. 
See z-dither branch. One of the versions of slice_volume in PrintObjectSlice.cpp contains commented out call to z_dither(..), which returns polygons for all five regions. In this commented out form z-dither branch performs exactly like the master branch. It is not clear what is the proper way to incorporate the algorithm into the rest of PrusaSlicer. I intended to bring the algorithm to the attention of core developers and start the conversation.

kubispe1, 2023-01-17:
I find it is simmilar to this issue. #5860 Thanks for your detailed description.

LRaiz, 2023-01-17:
If desired, I could open a pull request to integrate the z-dither branch into https://github.com/prusa3d/PrusaSlicer.git . Submitted code changes are very localized. They calculate all five z-dither components of layer composition. I think you may consider incorporating the z-dither branch because the invocation of z_dither(...) is commented out, so the existing slicer's behavior is not affected. Let me know if you would like me to open such pull request.

I hesitated to make any additional changes because they would affect more areas of the PrusaSlicer than I am prepared to touch. It was not clear to me what was a better approach to take. Replacing a single ExPolygon in a Layer with five ExPolygons seemed too drastic of a change. Introducing pointers to four additional optional sublayers (each with its own ExPolygon) into a Layer may be better. I will leave this decision to core developers.

LRaiz, 2023-04-19:
I ended up studying much of the existing codebase and finding a solution that was consistent with it. Therefore, my pull request represents the full implementation I had previously hesitated to make.

LRaiz, 2024-01-09:
The bid advantage of using z-dithering is an ability to print faster without loosing model details. This may be achieved by using z-dithering with larger extruder and simultaneously increasing layer height. E.G. 0.6 mm extruder nozzle with 0.3 layer height will print details as well if not better than 0.4 mm nozzle with 0.2 mm layer, but will do it 20% faster.

k1-801, 2026-09-18:
> **This is a final warning**. If no new user comment lands on this issue in the next week, it will be **automatically closed**.
> 

User comm3nt landed.
The request is still relevant.