prusa3d/PrusaSlicer #9389: Convert Part into Modifier Mesh

tuxmobil, 2023-01-12:
I would like to modify sections of functional parts to have different properties, like infill, to strengthen them where needed.
When the geometry is a bit more complicated than a cube, I found the easiest way so far to create the modifier mesh needed, is to already it in CAD and then load the combined part into PrusaSlicer. Then I would like to use the inner part as a modifier mesh.

This has the benefit that the modifier mesh is already in the correct position.

Currently I cannot convert the already loaded part into a modifier mesh. I can load an STL as modifier mesh, but then the position has to be adjusted manually which is a tedious job.

Suggested solution: right click on part -> convert to modifier mesh

YuSanka, 2023-01-13:
Do you know about context menu item "Change type"? ;)

neophyl, 2023-01-13:
Have you split your original imported stl into parts ?  Until you do this the individual sections cant be selected.

tuxmobil, 2023-01-13:
> Have you split your original imported stl into parts ? Until you do this the individual sections cant be selected.

Yes, that was part of the process. I should have mentioned that.

tuxmobil, 2023-01-13:
> Do you know about context menu item "Change type"? ;)

I have the feeling to have seen that in the past (or being mentioned), but I cannot find it in 2.5.0.

[image]

Where is it? :eyes:

neophyl, 2023-01-13:
I just tried it on windows v2.5 with a multipart stl.  SPlit it into parts, selected one of the parts, moved it down slightly to intersect the model more and right clicked it and selected Change Type, selected negative volume and sliced and it removed the part of the model it intersected with.  So the functionality does seem to work.

neophyl, 2023-01-13:
OK, try this, open the attached project.
[multipart_test_model.zip](https://github.com/prusa3d/PrusaSlicer/files/10410225/multipart_test_model.zip)

you should see one stl file in the tree.  Select the object and use the split to parts tool from the top tool bar.

The tree should now have 1 object with 2 sub parts in it.  Select 'multipart_test_model.stl_2.  This should highlight the cylinder part of the model.

Now right click that sub part and you should see a menu, that should have the option for change type.

tuxmobil, 2023-01-13:
Alright, thanks for your all your input!

My error was that I split the multipart STL into seperate objects instead of parts. While the "change type" option is available for parts, it isn't for objects. That's where I was stuck. It didn't came to my mind that this was a precondition for this feature to work. 

While technically it could be feasible to change objects into modifiers as well, I agree that this not necessary to achieve my initial goal, since it already is available if I take the right route from the beginning.

Objects loaded from disk can directly be loaded as modifiers, so this seems to be covered as well.

This issue is therefore obsolete. Thanks. ;-)

neophyl, 2023-01-13:
Just a note that when you load an object from disk as a modifier you are loading it as a PART of the first object.  You are not loading it as a separate object.
Anyway glad you got it figured out :)

closed 2023-01-14 by lukasmatena