prusa3d/PrusaSlicer #9391: GUI: Object Manipulation Panel blank/unusable. PrusaSlicer-2.5.0+linux-x64-GTK3-202209060725.AppImage

noisufnoc, 2023-01-13:
### Description of the bug

Using the AppImage GTK3 version of PrusaSlicer-2.5.0, the "Object Manipulation" panel is completely blank and unusable.  

### Project file & How to reproduce

Open PrusaSlicer 2.5.0 AppImage, review Object Manipulation panel.

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+linux-x64-GTK3-202209060725.AppImage

### Operating system

Ubuntu Budgie 22.10

### Printer model

Prusa i3 MK3

kubispe1, 2023-01-16:
Uff, we will try to reprodue it.
Thanks

kubispe1, 2023-01-17:
Ok, so it is reproducible only on **Ubuntu Budgie 22.10**. 
[image]

noisufnoc, 2023-02-01:
reproduced the error in version_2.6.0-alpha2 as well

fornellas, 2023-04-03:
Still an issue with 2.5.2 or 2.6.0-alpha6:

[image]

The issue seem to be related to all Widget themes Pocillo-*, any other theme I tried (eg: Arc-Dark) works fine:

[image]

So it seem the theme + Prusa Slicer combination is broken for... this panel only Oo.

Another alternative, is to use the GTK2 version:

[image]

kubispe1, 2023-04-05:
@fornellas, @noisufnoc thanks for the additional info. We collect these Linux issues related to GUI. But honestly, it has less priority on our tasklist, at the moment.

lukasmatena, 2023-10-23:
@noisufnoc Can you please retest it with 2.7.0-alpha1? Thanks.

noisufnoc, 2023-10-26:
Hello, unfortunately the issue still persists on 2.7.0-alpha1

[image]

polymachine, 2024-04-12:
I'm also experiencing this issue in PS 2.7.4 on Arch

chlo-ee, 2024-05-22:
Can confirm - this issue is present in 2.7.4 on Arch.

MaximilianBuegler, 2024-06-10:
I can also confirm with 2.7.4.1 on arch. Starting with a different GTK theme works though since it seems to be related to dark themes:

`GTK_THEME=Adwaita:light prusaslicer`

closed 2026-09-29 by prusaslicer-bot[bot]