prusa3d/PrusaSlicer #9393: Feature Request: Add option for print order "inner wall>outer wall>inner wall(s)>infill" for 3 or more perimeters

The760, 2023-01-14:
I've done a lot of experimenting with slicer settings to achieve better surface quality, mainly to combat issue 602 that plagues us all. The best solution I've found is a feature implemented in the SoftFever fork of BambuStudio.

PrusaSlicer already has a setting to print outer walls first before inner when using 2 or more perimeters, which does typically lead to better surface quality in my findings. However, this can be detrimental to overhangs, and leads to seams being a lot uglier. When using an order of inner>outer>inner(s)>infill though, this issue becomes much improved, due to one of the inner walls being first.

Implementing the order of inner>outer>inner(s)>infill could be very beneficial to people who are unhappy with their print quality, as it has definitely improved all of my results. Prints have more consistent extrusion, especially when held under harsh light, and are smoother to touch. This is greatly beneficial for parts I print for work that customers press with their fingers.

If I had to guess why it helps, it's because the outer wall gets to print without anything behind it, and gets more dedicated time to cool. This setting has also helped slightly minimize the Benchy Hull Line issue, which is directly related to the thermal expansion of filament ( and I suspect it plays a factor in 602).

Would really love to see this implemented if at all possible. My two MK3S's are going strong still, and with this, they could be producing even better quality parts.

stefanbayer1, 2023-11-10:
Any news on this implementation request- will this eventually make it to the implementation backlog, when comparing round objects and the seam lines Prusa MK4s Input Shaper quality is definitely behind Bambulabs...
Would really love it to see this setting implemented.

ctadlock, 2023-12-20:
Looks like this was implemented in OrcaSlicer.

[image]

dohlin, 2024-04-19:
Bump!! Please add this!

bpwmd, 2024-05-22:
It already exists under layers and perimeters

[image]

Trecoolius, 2024-05-30:
Bumpity bump.

Widsa2024, 2024-06-01:
Bump, yes please.

TRONJon, 2024-06-09:
I'd love this!

barcoder20, 2024-07-28:
Bump, this would be very beneficial!

Mental-Inferno, 2024-09-28:
BUMP!

This feature is amazing, especially in high speed printing. You can set the outer perimeter speed for quality and set the inner perimeters as fast as your hotend allows. The initial print of the inner perimeter primes the nozzle for a great outer perimeter while still allowing for the rest of the perimeters to be laid out quickly and without affecting dimensional accuracy.

Racemade, 2024-09-28:
+ would be great!

Mental-Inferno, 2024-10-12:
Just wanted to note that this perimeter pattern is also known as "Sandwich Perimeters".

IronFist16, 2024-11-08:
Is this feature going to be implemented?

Racemade, 2024-11-13:
Up :)

Mike-Flexicon, 2024-12-13:
I'd really love this feature to be added.

Double077, 2025-01-14:
Why is this not added yet !

dboesten, 2025-02-16:
This would be very helpful for prints that involve placing bearings so the overhang to support the bearing prints more consistently

freedom14seven, 2025-03-02:
Would be a great feature! Here is a video showing the difference in OrcaSlicer where the option is already available: https://www.youtube.com/watch?v=__OQmUwVkrw

christhabot, 2025-04-05:
there's an option called "external perimeters first" in layers and perimeters -> advanced, this isn't as good as inner-outer-inner but its better than nothing if you have the problem of perimeters getting pushed outwards

coho11111, 2025-04-09:
BUMP.  How is this not on prusaslicer

dkbay, 2025-05-01:
This is one of the features I use on basically every print I sell in Orcaslicer.
It would be really nice if this could also be added to Prusaslicer!

willmackey2nd, 2026-08-20:
At least for my models I don't see any reason for this to be the DEFAULT as it gives the best of the worlds for most stuff.
External first is useless if there are any overhangs as it just give poor performance for those sections.

Epsylon42, 2026-09-10:
As far as I know, this is still not implemented in the 3.0 branch, so commenting to keep it open