prusa3d/PrusaSlicer #9394: Ramming Customization Window Buttons Cut Off

Spencer-Kessler, 2023-01-15:
### Description of the bug

After updating to 2.5.0 and while setting up some new filament profiles, I noticed the buttons which allow you to customize the ramming time and other parameters in the ramming customization window are cut off. This is especially a problem for the total ramming time since you can only adjust the value by using the buttons, meaning the value cannot be increased, only decreased.

### Project file & How to reproduce

[project_file.zip](https://github.com/prusa3d/PrusaSlicer/files/10419285/project_file.zip)
[image]
[config.zip](https://github.com/prusa3d/PrusaSlicer/files/10419295/config.zip)

Steps to reproduce:

1. Go to "Filament Settings"
2. Got to "Advanced"
3. Scroll to bottom and click the button next to "Ramming parameters"
4. The window will open and only the minus buttons will be visible, with the plus buttons cut off

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+linux-x64-GTK3

### Operating system

Ubuntu 20.10

### Printer model

Prusa MK3S+ w/ MMU2S

kubispe1, 2023-01-17:
There are few places where we discoverd simmilar issue. It is point to GTK3 in 2.5.0 and probably will persist in 2.6.0 for Ubuntu.
Thanks and have a nice day.

NyxCode, 2023-02-06:
For example, it'd be great if the value could be entered by keyboard. That would be a workaround for the cut-of UI elements.

lukasmatena, 2023-10-20:
Closing as a duplicate of #6399. It will be fixed in the upcoming release.

closed 2023-10-20 by lukasmatena