prusa3d/PrusaSlicer #9395: update github

iplayfast, 2023-01-15:
I've been waiting  for new features as described in your videos to appear (tree supports), but nothing has been updating in the source code for months. 
I feel like development is happening behind the scenes and not in the public. Is there a tag that development is happening on. 

If so, could you update the readme to say where current development is happening?

RedCoder09, 2023-01-16:
I believe all active development is happening privately, unfortunately, so I don't think we'll get any updates until 2.6 alpha1.

Area5142, 2023-01-16:
Also noticed that github has not been updated in recent months - unfortunately.
This is a new direction for Prusa, previously you could find the latest development, even pre-alpha versions of PrusaSlicer and compile the code yourself.

It is, in my opinion, a step backwards for the open-source idea and for community contributions.

Maybe it is linked to the upcoming release of Prusa XL and the support of this printer in PrusaSlicer, which is kept secret until it is released.

Snuff1eupagus, 2023-01-17:
I've had a bad feeling about this since the alpha's of 2.5, and the 2.5.0 final release was very poor in my opinion. The company has to do what it's got to do, this could mean abandoning the community aspect, remember Makerbot.

gluap, 2023-01-17:
I'd like to second that - I'd very much like to see development in the open. It's very frustrating as an open source developer to see features announced on youtube just to find that they somehow haven't been committed to github yet (or not in a public repo anyhow) - If development has to be in secret for some reason, I'd prefer not to be hyped about the hidden development on other channels.

etet100, 2023-01-20:
In my opinion, it is related to competitive projects. For example, Bambu Studio. They don't want these new features to come to competitors before version 2.6 comes out.

Alexx40in, 2023-01-20:
> In my opinion, it is related to competitive projects. For example, Bambu Studio. They don't want these new features to come to competitors before version 2.6 comes out.

Most of these new features have been in Bambг Studio for a long time :)

gudvinr, 2023-01-20:
> Most of these new features have been in Bambг Studio for a long time :)

Can you list all of those new features that were teased that bambu slicer has "for a long time"? Not a lot of stuff was announced so, one or two features while _technically_ is the most, it is not a significant amount. And this is only what was shared anyway.

Not to mention that sole existence of bambu slicer in this shape and form was possible because PS and Slic3r did all the heavy lifting anyway.

Alexx40in, 2023-01-20:
> Can you list all of those new features that were teased that bambu slicer has "for a long time"? Not a lot of stuff was announced so, one or two features while _technically_ is the most, it is not a [...]

organic supports, 3d text, advanced model cutting

gudvinr, 2023-01-21:
So, as I said, it is not know how much of the stuff is going to be released, so calling "most" based only on 3 is kinda overstatement.  
At least 2 of these were released in the middle of developing cycle between 2.5.0 and 2.6.0, not to mention that first version of BS was released less than a year ago, so "for a long time" is a stretch too.

They did introduce STEP import first, for example. Multi-color workflow in BS is outstanding too. Let's give them credit where credit's due. But many things they do, they happily take from other projects, including new features they have "for a long time".  
It's not bad per se because open source and stuff, but it definitely hurts one company when other company releases first company's work faster. Open source or not, but business is business.

Heck, compare [their](https://bambulab.com/en/download/studio) product page vs [prusas](https://www.prusa3d.com/page/prusaslicer_424/).  
Bambu has _zero_ mentions of what they use to make it possible. "cutting edge" my ass.  
And PS not only list being based on Slic3r as their _first_ feature, they still have Slic3r in their window title.

faith-healer, 2023-01-28:
Saturday.

Snuff1eupagus, 2023-01-29:
This is exactly the type of thing I don't expect Newbies to understand. and why I think they should be removed from the suggestions relative to the software development, newb's that don't know there ass from a hole in the ground are being invited to suggest improvements to the software and they are doing more harm then good. persons not knowing how to implement the software should be removed from the conversion, as they don't have a clue whats needed for the proper development. 
Now we are seeing tools implemented that  don't take into consideration the capabilities that we already posses. not only that. we are seeing the development of tools for specific tasks rather then tools designed to give us the ability to create tools, as needed. I would use Photoshop and GIMP as an example of this. They provide us with the ability to build what we need. PrusaSlicer is being designed for those that don't  wish to learn how to do what they need to know, but rather for those that don't know and don't want to know.  education should be the focus not just spending Dollar after Dollar developing software to combat the new users that can't be bothered to take the time to learn how to use it. 
I'm very sad where I see this going. 

Sorry new users ,you will never know, you will simply move to a new format thinking this is a waste of your time. probably for the best.

gluap, 2023-01-29:
@Snuff1eupagus you seem to have felt the need to vent a bit, but to I wonder you mean to say by your comment. You seem aggravated that somehow you think PrusaSlicer is going down the drain, but how you think that follows from or relates to "infrequent source code repo updates", which is the topic of this issue, remains unclear.

I also don't really see how Gimp and Photoshop relate to the issue at hand (with Photoshop not having any open sources and Gimp pushing almost daily).

bubnikv, 2023-02-08:
Alpha released, closing.

closed 2023-02-08 by bubnikv