prusa3d/PrusaSlicer #9396: Small Perimeter Feature Length

derekonilsen, 2023-01-16:
**Is your feature request related to a problem? Please describe.**
My printer doesn't do well with the default small perimeter speed, however, when attempting to use SuperSlicer which offers this feature, my printer profile is aweful in comparison, I feel this solution would not affect the default  Prusa Slicer features while also increasing quality on printers which need slower speeds for perimeters of certain lengths. 

**Describe the solution you'd like**
Implement a variable which user inputs for small perimeters. 

**Describe how it would work**
Simply implement a variable which user inputs for small perimeters length that calculates length of perimters and ensures that perimeters small than input length are at lower speed. 

**Describe alternatives you've considered**
Super slicer, and lowering perimeter and external perimeter speeds, but for prints which require more perimeters and have large sections as well as small holes and such, it becomes impossible to achieve speeds necessary for fast and precise printing. 

**Additional context**
Above details enough.

CCS86, 2023-02-18:
Agreed, the small perimeter speed can be super disruptive and there is currently no way to turn it off. 

By adding a "max small perimeter length", we could tune when it is used, or disable entirely.

hessmann920, 2024-03-18:
I just ran into this limitation while tuning a profile for my machine. 

Adding a new parameter to define when to implement Small Perimeter settings would be extremely helpful. The machine can slow down for extremely fine details without disrupting the entire print. 

I am setting up a Prusa configuration profile for my Peopoly Moai, and the supports for small models requires a significant slowdown of the laser to increase exposure of the support tips. Otherwise, the supports are not strong enough to pull the model off the vat. 

Unfortunately, the hardcoded implementation of this small perimeters function takes a print from nearly 9 hours to over 14. See the picture below showing the "small Perimeter" speeds applied to a significant portion of the model.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]