prusa3d/PrusaSlicer #9397: Preview mode add option "auto refresh"

livius2, 2023-01-16:
**Is your feature request related to a problem? Please describe.**

When i change any parameter e.g. fill percentage, fill pattern, number of perimeters... I must press slice now every time.

**Describe the solution you'd like**
**Describe how it would work**

Please add checkbox to "auto refresh/update (slice)" on every change. 
When checkbox is checked and i change parameter, it automaticcally reslice the model (no need to click on "slice now").
This is annoying when you prepare model and you experiment with values and must every time click "slice now".

It is enough even if the option will not be saved anywhere in the settings, only used ad hoc.

[image]

YuSanka, 2023-01-16:
Thanks for your report. But this possibility already exists ;)
[image]

kubispe1, 2023-01-17:
Thanks, I am closing this issue..

closed 2023-01-17 by kubispe1