prusa3d/PrusaSlicer #9398: [Feature request] Shape gallery dynamic text recycle symbols.

GizmoTheGreen, 2023-01-17:
**Is your feature request related to a problem? Please describe.**
I had a real AHA moment when I discovered the shape gallery. since it's easy to add an emboss/deboss symbol to remember what material you printed the part in, for when/if you throw it away.

but there is only PLA, PETG and OTHER, while we have so many more plastics to 3d print with.
and even then it would take some effort to replace the model with the new one each time you reslice for a new filament.

**Describe the solution you'd like**
The text part should be dynamically generated depending on the currently selected Filaments Filament type (under advanced).

Now you'd only have to add the logo once (position, rotation, size).
if you first printed it in PLA but now want to print in Nylon or PCTG or TPU, simply select the appropriate filament profile and the model will dynamically change/rebuild to the correct Filament Text in the recycle icon.

foreachthing, 2023-01-17:
I made this https://github.com/foreachthing/RecyclingSymbols a while back. Feel free to use/add/mod.

foreachthing, 2023-01-18:
Not as dynamic as you wish for, but I made a bunch of recycling symbols ready to import here: https://github.com/foreachthing/RecyclingSymbols

Question: what would happen, if you mod your model? You'd still have to re-place the symbol, right?
So, it should be easy enough to just replace the current symbol with a new one, if you should change your material.

murk-sy, 2023-01-18:
See also/Possibly duplicate: #4023 

Ideally, embossing/debossing should support variables similarly to how they work with the gcode filename.

GizmoTheGreen, 2023-01-18:
> Not as dynamic as you wish for, but I made a bunch of recycling symbols ready to import here: https://github.com/foreachthing/RecyclingSymbols
>
> Question: what would happen, if you mod your model? [...]

Thanks, I'll give those a look in the meantime.

Answer: depends if you mod your model. in my case I might just want to print different plastics to compare strenght, or maybe I only have a certain color in another type than what I originally sliced.

if you did modify the model in such a way you'd need to reposition it then you might as well start over with the symbol, I guess. 
but it also would be faster if it replaced/regenerated in place.

------------------------------------------------

I mainly suggested this feature cause they had a set of symbols in the gallery already, and dynamically generating the text would future proof it! (no need to wait for new symbol models when a new filament type hits market)

closed 2026-09-29 by prusaslicer-bot[bot]