prusa3d/PrusaSlicer #9399: 0.4 layer + 5 layers of raft + lightning infill = crash

Pitel, 2023-01-17:
### Description of the bug

Crash. I expected it to, um, well, not crash.

### Project file & How to reproduce

1. Have Prusa MINI with 0,6 nozzle profile
3. Add default cube (or probably any other STL or anything)
3. Choose any layer height larger than 0.2
4. Set infill type to Ligtning
5. Set raft to 5 layers
6. Crash 💥

You can do steps 3-5 in any order, but after the last step, it always crashes.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Verze 2.5.0+linux-x64-GTK3

### Operating system

Ubuntu 22.10

### Printer model

MINI

rtyr, 2023-01-17:
Reproduced, thanks.

[Lightning-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/10435591/Lightning-Box.zip)

bubnikv, 2023-01-17:
Fixed with 402affb03521d0dbc159f98c24a5964edc23715b
It will be fixed in first PrusaSlicer 2.6.0 public alpha.

closed 2023-01-17 by bubnikv