prusa3d/PrusaSlicer #9402: no filement

comemilg, 2023-01-17:
### Description of the bug

my problem is when I slice it says that it going to use no filament. and when I print no filament comes out. i have search on the whole internet and find nothing.

i have designed my own ting in fusion 360 and exported it.
and tried to download from thingiverse.

### Project file & How to reproduce

[bult 10mm sliced stl zip.zip](https://github.com/prusa3d/PrusaSlicer/files/10439279/bult.10mm.sliced.stl.zip.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2,5,0

### Operating system

windows 11

### Printer model

cz-300 Ctazy3DPrinter

kubispe1, 2023-01-17:
I can slice your model without any trouble. The filament amount looks ok ... 
[image]

comemilg, 2023-01-18:
Do you know if I have forgot some settings because it’s my fist print on any slicer ?

because when you just sliced it you got filement use ?

neophyl, 2023-01-18:
Can you please attach a Prusa Slicer 3mf project file instead of a stl.  Your stl slices fine.  It will be something in your settings which a project file contains. With a project we can examine all your settings.

Best guess the most likely culprit is your G-code flavour, located in Printer Settings>General>Firmware.  I have seen users have '|No Extrusion' selected in the dropdown which then behaves as you describe as in effect you are configuring for no extrusion, none is performed.
Thats just a guess though and a project is required to give a more reasonable answer.

comemilg, 2023-01-18:
thank you noephyl you are the best 👍
You saved me from even more headache. 

and thanks to kubispe1 for helping as well.