prusa3d/PrusaSlicer #9403: Unable to export generated g-code

psla, 2023-01-18:
### Description of the bug

I clicked "slice", then I clicked "export gcode". I am presented with following error:

```
Failed processing of the output_filename_format template.
Parsing error at line 1: Not a variable name
{input_filename_base}_{layer_height}mm_{initial_filament_type}_{printer_model}_{print_time}.gcode
                                        ^
```

### Project file & How to reproduce

n/a

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.0

### Operating system

Winows 11

### Printer model

2.4.0+amd64

psla, 2023-01-18:
FWIW I tried the same 3mf file on a different computer with 2.5.0 slicer and it worked fine.

kubispe1, 2023-01-18:
Hi @psla, 
`{input_filename_base}_{layer_height}mm_{initial_filament_type}_{printer_model}_{print_time}.gcode`
I checked and, just in case, put this statement into the output file format field and that looks ok. Attach your 3mf, if your issue persists.

bubnikv, 2023-03-20:
initial_filament_type was introduced in 
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.1-beta1
thus it will not work with PrusaSlicer 2.4.0.

closed 2023-03-20 by bubnikv