prusa3d/PrusaSlicer #9405: Pressing control in g-code upload dialog moves cursor to the end of the text box

Adrian-at-CrimsonAuzre, 2023-01-19:
### Description of the bug

As the title says, pressing the control key when the upload g-code textbox if focused moves the cursor to the end of the line. I can not replicate on Windows, so I suspect this is a Linux/AppImage bug.
[image]

### Project file & How to reproduce

1. Have PrusaSlicer running via AppImage (Ubuntu 20.04 for me)
2. Have a printer with Octoprint, Mainsail, Prusalink, etc set up
3. Slice anything
4. Press upload G-code
5. While editing the file path, press control.

This renders copy/paste or control+backspace useless.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0

### Operating system

Ubuntu 20.04.5 LTS

### Printer model

Prusa MK3S

Jan-Soustruznik, 2023-01-20:
Hello, @Adrian-at-CrimsonAuzre , 

I have the same OS, Ubuntu 20.04.5 LTS, but I cannot reproduce your reported issue. Maybe I didn't understood correctly .. Could you more specify your issue? Thank you, have a nice day.

Adrian-at-CrimsonAuzre, 2023-01-21:
@Jan-Soustruznik If I highlight the word "Carriage" and press control, my cursor shoots to the end of the text box.

I've actually just discovered that any time the window loses focus (like alt+tabbing or pressing print screen) the cursor position in this box _also_ resets to the end of the line.

closed 2026-09-29 by prusaslicer-bot[bot]