prusa3d/PrusaSlicer #9408: Add Averages like Volumetric Flow Rate and Speed to the Sliced Info Section

D3MZ, 2023-01-20:
**Is your feature request related to a problem? Please describe.**
In both tuning and diagnosing issues with a certain brand of filament / printer, it helps knowing these heuristics. I find myself manually calculating this all the time, and this seems like a very quick addition as the numbers are already there. 

**Describe the solution you'd like**
On the Export tab, under "Sliced info" (just above the Export to G-code) please add the averages for g/s, mm/s, and mm^3/s please! 

**Describe alternatives you've considered**
I currently just manually divide the from the estimates that the slicer generates.

**Additional context**
Add any other context or paint drawing about the feature request here.

kubispe1, 2023-01-20:
Hi, please can you explain purpose for the required extra information? So you mean this simple calculation: Used Filament [g] / total print time [s], and so on for other cases...??

D3MZ, 2023-01-21:
@kubispe1 - yes that's it!

closed 2026-09-29 by prusaslicer-bot[bot]