prusa3d/PrusaSlicer #9419: Timelapse Mode/Settings

Alexakazam, 2023-01-23:
Requesting a timelapse feature be added.
I use a selfie stick remote shutter button attached to my mini+ using a custom bracket, the button gets pressed by the extruder head at X180 Y180 after each layer change to trigger an iphone to take a photo. The main issue I run into is on larger prints the remote will turn off due to being idle for too long. I try to click it between layers to keep it alive but sometimes I can't always get to it to do that. I would like a feature where you can set a duration of time to pass by, then force the extruder head to return home and press the button, even in the middle of a layer. When it decides to go home should be optimized so it isn't during the middle of laying down a line, but at the next or even previous best possible moment just to keep the remote alive. I know I am not the only person using this type of method to achieve this type of function. Just think it would be a very cool feature for us makers out there that like to record their prints or create youtube content.

mjfsch, 2023-01-24:
i wouldn't be surprised if this winds up quite low down on the todo list, but you can probably implement a solution for this using the post processing scripts feature - output options under the print settings.

Tupson444, 2023-01-24:
As @mjfsch suggested, you can use post-processing scripts. I don't know how to make a script, but I have an idea how to do this in another way:

Turn on "Supports remaining times" (under Printer Settings > General), so the GCODE would be filled with M73. Then you can use find&replace (after creating gcode, using a text editor, because it doesn't seem to work with G-code substitutions inside PrusaSlicer) to insert whatever you need at certain time intervals.

For example:
find "M73 P30" and replace* with "YOUR CUSTOM GCODE\nM73 P30" **
find "M73 P60" and replace* with "YOUR CUSTOM GCODE\nM73 P60" **
... and so on at whatever time interval you need
(For long prints and short time intervals it can be a lot of manual work, but I don't have any other idea. If a post-processing script can make this more automatic, e.g. to replace every first instance of M73 at intervals of 30, it would be easier.)

*Make sure that you don't choose "replace all", because the same M73 code would occur multiple times, and you want the custom code to activate only once in every, say 30 minutes.

**You can use Notepad++ to be able to insert "\n" (new line), because the M73 code also has "R" value which you don't want to end up in the same line as the custom code.

Alexakazam, 2023-01-24:
> Turn on "Supports remaining times" (under Printer Settings > General), so the GCODE would be filled with M73. Then you can use find&replace (after creating gcode, using a text editor, because it [...]

As of right now, only the MK3 supports this feature according to Prusa Slicer. Seems like this could be a potential option for me though if it was supported by my Mini+. It took me a second to understand your reply but after re-reading and thinking about it, it does kind of make sense what is happening here.

> i wouldn't be surprised if this winds up quite low down on the todo list, but you can probably implement a solution for this using the post processing scripts feature - output options under the print settings.

Yeah I don't doubt that, but it is still a feature I would like to see implemented into the slicer. I'm seeing more and more "headless timelapse" videos and think it would be great to have a little more control over it natively in the slicer. Maybe a better option for me would be to use a different button system, using the TRS plug method, and not some bluetooth remote off a selfie stick lol

Thanks for the replies!

Tupson444, 2023-01-24:
> As of right now, only the MK3 supports this feature...

It doesn't matter, because you would use the M73 code for other purposes. I tried turning on "show remaining times" in Marlin firmware on one of my printers (old Prusa MK2), but I gave up. However, the settings in PrusaSlicer remained as I may turn it on in future, so all of my gcodes have M73 commands included, even though I don't use them. As far as I know M73 can't cause any harm if the printer doesn't support it.

I agree that it would be a useful feature, as it would be much easier to do if there was simply an option to enter custom gcode at every X seconds. This could be used for other purposes potentially (whatever someone needs the printer to do at certain time intervals).

Alexakazam, 2023-01-24:
> It doesn't matter, because you would use the M73 code for other purposes. I tried turning on "show remaining times" in Marlin firmware on one of my printers (old Prusa MK2), but I gave up. However, [...]

I see. You're sort of speaking Greek to me but I think I'm picking up what you're putting down. 

> I agree that it would be a useful feature, as it would be much easier to do if there was simply an option to enter custom gcode at every X seconds. This could be used for other purposes potentially (whatever someone needs the printer to do at certain time intervals).

Indeed. Just an idea I had and figured I would throw it out there.

Tupson444, 2023-01-24:
As a summary, use "Supports remaining times" (no matter if the printer actually supports it) and then manually edit the gcode by replacing M73 commands.

When I saw this issue I thought of a workaround as described above. Someone with more knowledge may think of something better/easier.

Alexakazam, 2023-01-24:
Yeah this is quite a bit more advanced than I really am capable I think. Could you maybe explain your gcode example in your original reply a little more? Like the purpose and outcome of each syntax and would the custom gcode be the gcode I'm already using after layer changes, it would just happen during M73 instead? I'm still trying to understand how this works, not a native gcode speaker lol.

Alexakazam, 2023-01-24:
My honest thought at this point is to rip the remote apart and wire an arduino to jump the shutter button at a set interval. That way it still takes a photo after each layer and after every 9m59s, which may produce the occasional photo that could get taken during or after the move to X180 Y180 where the photo gets taken at, if the 9m59s trigger falls during that move in the gcode. But a small sacrifice to pay in post I guess.

Tupson444, 2023-01-24:
Ok, I'll try my best to explain it clearly.

First of all, I made a mistake, the "M73 P30 R14" means that 30% of the print is done, and that 14 minutes are remaining (not 30 minutes elapsed, oops). This makes it more difficult, and we need to do it more manually.

Here are the steps:

1. Turn on "Supports remaining times" and create a gcode file (slice the model) regularly.

2. Open the gcode file with a text editor (Wordpad, Notepad++, whatever that opens gcode correctly) and look for the first place where M73 shows up (should be very near the top of the file).

4. Take a look at the number after "R" (in the same line as M73). If it says, for example, "M73 P0 R80" it means that the whole file will print in 80 minutes.
Now we need to decide how often we want the custom gcode to activate. Note that the "80 minutes" is just the slicer estimate, not actual elapsed time. If, in your example, the remote turns off in 10 minutes, I would count down from the 80 by 9 (slightly less than 10, just in case time estimate isn't precise).

5. In the text editor, press ctrl+F and find: M73
Press Enter multiple times ("find next") until you reach a place where it says R71.

6. Right below that (create a new line) type your custom gcode, for example:
G1 X180 Y180
(or whatever you use after layer change)

7. Repeat steps 5 and 6, only keep counting down (e.g. R62, R53, R44, R35 ...).

Note that when you reach the same number multiple times (for example R71 a second time), don't type custom gcode again (you only want it once at 71 minutes remaining).

-------

You can remove your custom gcode after layer change (I guess you set it in the slicer, under Printer Settings > Custom G-code) if you want it to take a photo only after each X minutes. Or you can keep the code so the camera would be triggered at every layer change AND after every ~9 minutes.

I haven't tested this, but I hope it will be helpful.

So this is quite a manual work, and I hope someone else can help here to think of a better (more practical) way.

Alexakazam, 2023-01-25:
Wow, thank you for this. This now makes perfect sense. I will look into it and see if it does end up working for me and report back.

Alexakazam, 2023-01-25:
Well, technically that worked. I'm really starting to also question my remote. Set to one minute intervals it still turned off for some reason but was working when I started it. Also my custom gcode is very crude and it did exactly what it was told, issue with that is that it dragged across the z axis without lifting, also didn't retract so it was very stringy. Also, incredibly slow compared to doing it after a layer change for some reason. Lots of other issues associated with that method, mostly in the gcode realm. Some better custom code would probably make that work effectively but I wouldn't know the first thing about coding that gcode manually.

Tupson444, 2023-01-25:
Here is a gcode that you can use:

> G91
G1 E-1 F1800
G1 Z0.3
G90
G1 X180 Y180 F3600
G91
G1 Z-0.3
G1 E1 F1800
G90

expalantions:

G91          <----- set to relative positioning - important to go before the next two commands, to make the Z move up by 0.3mm and not down close to the bed, and for retraction to work properly.
G1 E-1 F1800     <----- E-1 is retraction length set to 1mm, F1800 = 1800mm/minute = 30mm/s retraction speed. Change these values if you use different settings. I'm not sure if this will work well, it might depend on other machine settings that I don't know about. You must test it to see if the retraction happens as expected.
G1 Z0.3    <----- change number if you need different Z raise
G90          <----- set to absolute positioning - again important to go after the previous commands, in order for the following commands to work properly
G1 X180 Y180 F3600  <----- F3600 = 60mm/s, change it if you need it to be slower or faster

The following commands will make the nozzle return to the original height by lowering 0.3mm. However, it will happen before the head returns to the printing (the whole process would go like this: retract, raise Z, move to X180 Y180, lower Z, deretract, return X & Y to continue printing). This means that the nozzle would drag across the print (and ooze) in return.
To solve this, when you find the place where you want to put the custom gcode, after M73, read the previous G1 command that has X and Y (above the M73, might be several lines above). It would look something like this:
> ...
G1 X125.193 Y62.288 E2.798
M73 P2 R14
...

Copy the G1 command without E value (e.g. only "G1 X125.193 Y62.288") and place it in your custom gcode before the following commands.

G91 <----- same as above
G1 Z-0.3 <----- if you change the Z raise number, you should change this as well (it must be the same, only negative)
G1 E1 F1800 <----- deretraction, change values if needed
G90 <----- same as above

So, the final custom gcode would look something like this:

> G91
G1 E-1 F1800
G1 Z0.3
G90
G1 X180 Y180 F3600
G1 X125.193 Y62.288 <---- change this
G91
G1 Z-0.3
G1 E1 F1800
G90

-------

For the between layers, you might also need to manually read the last G1 command above layer change (that has X and Y) and manually insert it in the position described above.

Or, as an alternative, you can find what are the X & Y coordinates in the center of the printed part, so you can place those coordinated in every place, instead of manually reading previous G1 commands. This would make the nozzle lower into the middle of the print, so it would drag and ooze only across the internal part of the print.

Xelinor, 2023-04-23:
> 

So, I'm trying to implement something similar to this for my printer, which automatically takes a picture after layer change, and this is what I've come up with:

```
{if spiral_vase == false}
G1 E-1 F1800
G0 Z{layer_z + 1}
G0 Y110 F18000
{endif}
;AFTER_LAYER_CHANGE
;LAYER:{layer_num + 1} Z:{layer_z}
{if spiral_vase == false}
G1 E1 F1800
G0 Z{layer_z - 1}
{endif}
```

However when I slice with that, it tries to bury the first few layers underneath the build plate? Any chance you can spot what I've done wrong?

Tupson444, 2023-04-23:
@Xelinor

Disclaimer: I might be wrong about anything here, proceed with caution.

You should remove the last part which lowers the Z, and put the code in "Before layer change G-code":

>{if spiral_vase == false}
>G1 E-1 F1800
>G0 Z{layer_z + 1}
>G0 Y110 F18000
>{endif}

and put this "After layer change G-code":

>{if spiral_vase == false}
>G1 E1 F1800
>{endif}

When you put the code before layer change, when the nozzle raises to layer_z + 1, which is say 0.3mm + 1mm = 1.3mm, then the the start of the next layer will instruct the printer to move to Z0.6 and all is good.

If you put layer_z - 1 afterwards, it will only try to go 0.6mm - 1mm = -0.4mm.

Regarding the unretraction (G1 E1 F1800), it would probably happen while it's still at the Y110, which means that it would ooze while returning to the print position. Also, the print head will move exactly at the layer height in the return (not raised 1mm). I don't have an idea how to solve this (other than manually changing Gcode which is impractical).

I haven't tested this, and as I said, I might be wrong.

Xelinor, 2023-04-23:
@Tupson444 

That certainly SOUNDS like it should work? Doing this this way at least makes sense, and it stops trying to render it inside the build plate, so that checks out. I'll give it a once-over and give it a try if it checks out, thank you!

closed 2026-09-29 by prusaslicer-bot[bot]