prusa3d/PrusaSlicer #9420: Internal error: Calling AppConfig::save() from a worker thread! Nvidia GPU error new driver.

ADI-Garage-CRedding, 2023-01-23:
### Description of the bug

When booting into PrusaSlicer, no matter the version, the below errors appear, **when I am using my dedicated GPU**, not my integrated, and only since firmware update 528.02, and still with 528.24
*Unhandled unknown exception; terminating the application.*
[image]
*Internal error: Calling AppConfig::save() from a worker thread!*
[image]
My dedicated GPU is an NVidia RTX A2000 Notebook and the Integrated is Intel UHD Graphics with driver 31.0.101.4032
I can set the application to run using the iGPU, however in the long term I'd like to use the dGPU
CPU, in case it is relevant: i7-11850H

### Project file & How to reproduce

Opening the program

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10 Enterprise 19042.2486

### Printer model

Prusa MINI+

neophyl, 2023-01-23:
Did you even search the existing issues ?  https://github.com/prusa3d/PrusaSlicer/issues/9178

ADI-Garage-CRedding, 2023-01-23:
> Did you even search the existing issues ? #9178

I did find both of the existing issues mentioning this before. Solution #9178 didn't work, I had to specify in windows settings to use the iGPU, which still doesn't fix the issues of using it with the RTX A2000 Notebook. Since the program is working fine with my RTX A4000 desktop GPU I'm wagering that the issue lies with Nvidia Optius and the Quadro series drivers and their interaction with API calls for Prusa Slicer. Solutions to #7839 also weren't applicable as much of that thread was based around virtual machines, something I'm not involved in my setup. I have found a patchwork solution however I wanted to document both that the issue exists and how to temporarily fix it for others having a similar issue.

kubispe1, 2023-01-23:
@ADI-Garage-CRedding and what about this hack ? Does it help to you? https://github.com/prusa3d/PrusaSlicer/issues/9370#issuecomment-1383049695

ADI-Garage-CRedding, 2023-01-23:
> @ADI-Garage-CRedding and what about this hack ? Does it help to you? [#9370 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/9370#issuecomment-1383049695)

I already have a temporary fix, I was hoping to get it permanently resolved and enable usage of my nvidia GPU with PrusaSlicer. I don't think this is something the community will be able to fix, this is an issue with the software itself.

gregbo24, 2023-01-31:
I'm seeing this same issue, similar notebook (Dell Precision 5570, windows 10 pro, A2000 card). I also am having problems with the software I use for 3d scanning (EXStar from Einscan) and I'm pretty sure the problems started at the same time, and I think my computer ran a GPU driver update and that's what caused it, but I tried to revert to the old driver and it didn't fix it. Something with the GPU update is causing some fighting to happen between the CPU, integrated graphics, and the GPU.

kubispe1, 2023-02-03:
Have you tried also 2.6.0? alfa 2? Please? Can you check ?

gregbo24, 2023-02-03:
Same result after updating to 2.6.0 alpha 2 a couple of days ago. Currently using the sw-renderer workaround to run 2.6.

bubnikv, 2023-02-03:
Do you guys run in any type of virtual machine? We have seen these reports
when running PrusaSlicer in Windows Sandbox, which is a virtualized
sandboxed Windows in Windows.

The error message is likely just a false alarm and we may disable it in the
future.

bubnikv, 2023-02-07:
Hopefully fixed with 84722876012ca310e29b291e10fef9d18ae26cea
Implementing a test whether the current thread is the main (UI) thread
and using it on AppConfig::save() to assert if save is called
from a worker thread.
The old assert was using thread names, which did not always work
on Windows.

lukasmatena, 2023-02-20:
Fixed, closing.

closed 2023-02-20 by lukasmatena