prusa3d/PrusaSlicer #9421: Extruder clearance in under print settings instead of printer settings

julie777, 2023-01-23:
### Description of the bug

It is hard to find the settings for the extruder that are used to prevent collisions during sequential printing. Because this is a physical characteristic of the printer, the setting should be part of the printer configuration and not part of the print settings.

If I change the printer being used in the slicer then the extruder settings need to change to make sure that there won't be a collision.

### Project file & How to reproduce

open the print settings output option

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

windows 10

### Printer model

Prusa MK3S+

julie777, 2023-01-23:
This creates a constant problem because I set my printer once.
I change print configurations regularly.

rtyr, 2023-01-23:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/589.

closed 2023-01-23 by rtyr