prusa3d/PrusaSlicer #9427: [FEATURE REQUEST] Pain-on Brims

mix579, 2023-01-25:
I can't believe I'm the first to request this but I searched the open issues and didn't find anything.

Currently, brims are all or nothing. Yes, you can use modifiers to turn them on and off for individual parts, but for any given part it's brim or no brim. And yes, you can use mouse ears to generate corner brims but many models would benefit from brims in just some selected areas.

There are undocumented techniques to accomplish this, e.g., https://forum.prusa3d.com/forum/prusaslicer/is-it-possible-to-have-a-selective-brim/

My request is to add an option for paint-on brims, identical in implementation to paint-on seams and paint-on supports.

neophyl, 2023-01-25:
You aren’t FUCHSR, I’ve seen a request previously.

mix579, 2023-01-25:
Thank you, please feel free to connect the requests; I guess my search skills are not up to par! 🙂

neophyl, 2023-01-25:
https://github.com/prusa3d/PrusaSlicer/issues/8283 And there’s at least 4 closed issue requests for the same thing 😊

mix579, 2023-01-25:
Okay, I clearly have no idea how github searches work ☹️ When I search for "is: issue is: open brim", I don't get a hit on #8283. When I search for "is: issue brim" I do get a hit, even though the issue is open. Anyway, doesn't really matter. I'll close this ticket . Thanks for pointing me to #8283

mix579, 2023-01-25:
Merge with  #8283

closed 2023-01-25 by mix579

Snuff1eupagus, 2024-04-26:
perhaps this will help you, this has been possible for several years, not paint on, however it might enlighten you as to was has been possible for a long time. Frankly this seems better then paint-on, as I have more control over the shape of my brims, and their application.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=4s

good luck,