prusa3d/PrusaSlicer #9429: Two profiles for Artillery Sidewinder X1/2

tom-321, 2023-01-26:
[image]

rtyr, 2023-01-26:
Why is that problem?

tom-321, 2023-01-26:
Hi @rtyr,
it isn't a problem. I've clicked on feature request.
Problem for me is is, I just want to remove one of the two Sidewinder's ('cause I have a BLtouch and mix them up)

neophyl, 2023-01-26:
Heres a workaround for you.  Copy the profile you want to keep and give it a custom name.  Then run the wizard to remove the printer.  Both of the system ones will be removed but you will still be left with your personalised copy.  Then just use that from now on.

tom-321, 2023-01-27:
Great 👍 Thanks

closed 2023-01-27 by tom-321