prusa3d/PrusaSlicer #9432: Additional change filament and stop mid layer

seifer69er, 2023-01-27:
This is a feature request.

**Describe the solution you'd like**
Pause of print and  filament change is triggered at layer change and can be added in prisaslicer.
However if a pause or filament change is needed mid layer print this is a manual gcode edit.

**Describe how it would work**
Add option to inject m600 pause or filament change during slicing so that manual gcode edit is not needed.
Recomment sliding the slicer to the point required then clicking add filament change just like it is done on the z axis of the slicer.

**Describe alternatives you've considered**
Manual edits work but should have the option available.

**Additional context**
Maninly required for single filament extruders as multi extruders wouldnt need this.
I have a prusa mini and find this super useful if only it can be added in the slicer.

murk-sy, 2023-01-27:
This is already possible - you can set up multiple extruders on a printer with just one and then use M600 as the tool change procedure.

Not sure if there's an official Prusa guide on this, but here's an article describing it in more detail: https://medium.com/@damien.martin.guillerez/multi-filament-print-with-a-single-extruder-using-prusaslicer-2e0746348cdd

rtyr, 2023-01-27:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5195.

closed 2023-01-27 by rtyr

seifer69er, 2023-01-27:
This issue was closed prematurely.
The link provded by @murk-sy is not valid for this feature.
[image]

seifer69er, 2023-01-27:
> Duplicate of #5195.

This is not a duplicate, this is independent of the link provided by [murk-sy](https://github.com/murk-sy)
See picture below

murk-sy, 2023-01-27:
Generally the way each individual layer is sliced is fairly automatic and although there are some controls available (perimeter printing order for example), putting gcode at a specific location would get lost immediately once the file is resliced by changing a setting.

Is there a particular reason why you would want to have filament change within the layer?

rtyr, 2023-01-27:
@seifer69er Look at https://github.com/prusa3d/PrusaSlicer/issues/5195. I believe it is the same request as you have.

> this is independent of the link provided by [murk-sy](https://github.com/murk-sy)

Marking this issue as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5195 is not related to any answers/links from murk-sy.

seifer69er, 2023-01-27:
> Generally the way each individual layer is sliced is fairly automatic and although there are some controls available (perimeter printing order for example), putting gcode at a specific location [...]

Using a single extruder, its difficult and labour intensive to manually edit gcode.
filament change within a layer can help produce different mechanical properties of the object being printed, with the added benefit of changing colour of the filament.
For example a semi rigid object - using both a conductive and non conductive filament to create a Printed circuit board.
The feature request in this thread is above and beyond that of #5195 as it provides the issue and the potential solution.

murk-sy, 2023-01-28:
I don't see how changing filament mid layer would improve anything. Maybe changing it specifically before infill may reduce the amount of purge required, but then you have to rely on purge being sufficient in the infill.
Like yes, using an automated height based finger joint of two materials would improve adhesion between them, but that's not really something you'd be doing by hand - but not sure if there's already a better

As for the circuit board, as I've said you don't really have fine control over what happens within in a layer. Using standard multi material printing and 2 separate objects (conductive separate from non-conductive) would be a LOT easier than essentially redoing all that work twice. Assuming each wire would be a separate "island" to have its own external perimeters, having them print in the correct order from non conductive parts (as to not have 15 manual filament changes in a single layer but just 2) would be just labour intensive. 

5195 and 5951 essentially cover what you've suggested so far, as far as I understand it.