prusa3d/PrusaSlicer #9435: "Multiply" could lay out the parts much denser

salamanders, 2023-01-27:
**Is your feature request related to a problem? Please describe.**
When I multiplied a triangle-ish part, they didn't fit on the build plate.

**Describe the solution you'd like**
Cura "interlaced" the triangles by alternating which direction each one was oriented.  (imagine a row of triangles facing towards the middle of the plate, then a mirrored row also facing inwards)

**Describe how it would work**
Hold out both hands with your fingers splayed.  Bring your hands together until your fingertips fit between your other hand's fingertips.  Like that!

**Describe alternatives you've considered**
Manually rearrange the parts after duplicating it.

**Additional context**
I was in cura for the skinny tree supports, but I hear I can do that soon in Prusa which is awesome!

neophyl, 2023-01-27:
It does this already if you want it to.  Right click on the arrange tool button and tick the Enable rotations option.  You can also set the spacing value in there.
[image]
 Gives the following with a wedge shaped part for example.
[image]

salamanders, 2023-01-30:
Fantastic, it worked great!  (Does it slow stuff down too much to be the default?)
I'm finding more and more hidden features, maybe they can pop up a "did you know, you can also rotate..." sort of guide.

closed 2023-01-30 by salamanders