prusa3d/PrusaSlicer #9436: Add a feature to save/restore EEPROM settings to the flash dialog

plampix, 2023-01-27:
Over at https://github.com/prusa3d/Prusa-Firmware/pull/3957 @wavexx mentioned there are tools to save/restore the EEPROM settings stored in your printer in case you would like to experiment/help with newer firmwares, and need a way to go back to your known-to-be-stable firmware without losing your settings. This would be a great addition to the PrusaSlicer firmware upgrade dialog.

wavexx, 2023-01-27:
I'm guilty. This can be done by using `D3` to first read, then again `D3 X` to restore the dump. After restoring a reset is advisable (so that the settings are forcefully read on restart).

cnlson, 2023-01-28:
If there's any way to genericize the back up to user selected settings (live z , sheets, nozzle) vs the whole eeprom. Might make it more compatible with restoring to dissimilar firmware

closed 2026-09-29 by prusaslicer-bot[bot]