prusa3d/PrusaSlicer #9437: Additional slicer option - change filament and pause mid layer

seifer69er, 2023-01-27:
This is a feature request.
Reopening as it was closed prematurely without understanding the feature.
Pelase do not quote"https://medium.com/@damien.martin.guillerez/multi-filament-print-with-a-single-extruder-using-prusaslicer-2e0746348cdd" - this link is irrelevant of the feature being requested.

Describe the solution you'd like
Pause of print and filament change is triggered at layer change and can be added in prisaslicer.
However if a pause or filament change is needed mid layer print this is a manual gcode edit.
The intetnion is to be able to add gcode edits in the slicer opposed to the gcode editor and trawling through the relevant line and manually editiing with more control and WITHOUT relying on multiple extuders.

Describe how it would work
Add option to inject m600, g04, pause or filament change during slicing so that manual gcode edit is not needed.
Recomment sliding the slicer to the point required then clicking add filament change just like it is done on the z axis of the slicer.
[image]

Describe alternatives you've considered
Manual edits work but should have the option available.
This link is invalid, and does not capture the feature being requested."https://medium.com/@damien.martin.guillerez/multi-filament-print-with-a-single-extruder-using-prusaslicer-2e0746348cdd"

Additional context
Maninly required for single filament extruders as multi extruders wouldnt need this.
I have a prusa mini and find this super useful if only it can be added in the slicer.

rtyr, 2023-01-28:
Duplicate of  #5195, #5951.

Please do not open another issue on this topic. You can comment on existing opened issues.

Closing.

closed 2023-01-28 by rtyr

rtyr, 2023-01-28:
BTW. And again, the link from murk-sy is indeed not related to your request, but it is also not related to the reason, why was your issue closed.