prusa3d/PrusaSlicer #9438: "411 Length Required" error on attempted network print

dfandrich, 2023-01-28:
### Description of the bug

I received a baffling *HTTP 411: 411 Length Required* error pop-up when attempting to print from PrusaSlicer to a Prusa Mini with the fancy new firmware 4.4.0 (also occurs on 4.4.1). I eventually tracked the problem to the *API Key / Password* field in the *Printer Settings* tab. I had cut-and-pasted the API key in to the field, but it included an embedded NL character (hex 0x0a). When I removed that character, it started working.

Clearly, PrusaSlicer is using the value of that field verbatim when sending the HTTP X-Api-Key: header, which adds an extra NL that, in addition to the NL sent by PrusaSlicer, terminates the HTTP header block before the Content-Length: header is sent.

PrusaSlicer should not allow its HTTP communication to be broken through bad user input. That field should be sanitized and illegal characters removed before use.

### Project file & How to reproduce

1. printf 'MyApIkEy\n' | xclip -i
2. Use middle mouse button to insert the key into the *API Key / Password* field. Note the arrow character (↵) at the end of the API key. In retrospect, that should have been a giveaway that something was wrong, but I just chalked it up to an odd UI choice and promptly forgot about it.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Linux

### Printer model

Prusa Mini

JulianGmp, 2024-11-09:
oh man to think I was troubleshooting for like 15 solid minutes

I don't know if control characters such as newline are considered valid characters for the API key, though I imagine some printers may support it.

Could I suggest to maybe add a warning to the user that when there are whitespace characters in the password/key?

JulianGmp, 2024-11-09:
I looked into this a bit more and inspected a wireshark recording of the request and I would actually classify this as a bug.

Since the API is key is sent as a field in the HTTP request header, it's terminated with `\r\n`. The additional newline that accidentally gets copy n pasted into prusaslicer ends up in the HTTP header like this:
```http
PUT /api/v1/files/usb//Shape-Box_0.8n_0.55mm_PLA_MK4IS_15m.bgcode HTTP/1.1
Host: prusa-mk4
User-Agent: PrusaSlicer/2.8.1
Accept: */*
X-Api-Key: REDACTED

Content-Type: text/x.gcode
Overwrite: ?1
Content-Length: 166158
Expect: 100-continue

```

An HTTP request however is terminated with an empty line (like after the `Expect` entry), so the server (my Mk4 in this case) doesn't actually process the fields afterwards (`Content-Type`, `Overwrite`, `Content-Length` and `Expect`).
Wireshark actually flags this as "Excess data after a body".

I would suggest that prusa slicer checks that the API key is a valid [HTTP token as of RFC 9110](https://httpwg.org/specs/rfc9110.html#tokens), since right now it also allows other characters that aren't valid as per the spec.

dfandrich, 2026-09-02:
I just tried this on 2.9.6 and Wireshark shows the same issue exists. The error code from the printer is now 408 instead of 411, but that printer firmware has also been upgraded since this bug was opened.