prusa3d/PrusaSlicer #9439: Logo embossing

Mike01Hu, 2023-01-28:
This is an adjustment to the text embossing implemented in 2.6.0 Alpha. 

It would be useful to emboss a scalable logo onto any surface, as is done with text embossing. The logo could be based on an enforcer model, .dxf file,  a STL or an exported surface file. This use would enable personalisation of generic models used in OEM environments, such as in casting moldings. This feature would save a lot of CAD time.

Of course, if this can already be achieved, clearly documenting the process would be appreciated.

0x6d64, 2023-01-30:
Is this different from adding STL files to the gallery and then adding them as negative bodies? There are already logos there, e.g. the recycling symbols for different kind of plastics. I have the feeling that your feature request is already implemented.

Mike01Hu, 2023-01-30:
> Thanks Martin, I'll investigate. Sometimes it pays to read the documentation!!!😉

jokohoko, 2023-01-30:
I think the main difference i see is applying the decal/embossing to a curved surface instead of a fixed geometry item from the gallery. Optionally mapping a model to a face would be great!

kubispe1, 2023-01-31:
Does Shape Gallery match your idea?
We would like to improve the upcoming Emboss function, but probably not in this way at the moment.

Mike01Hu, 2023-01-31:
> 

I will investigate shortly and update this thread when I have done so.

Mike01Hu, 2023-01-31:
Hi,

The shape gallery works on surfaces parallel to the base plate but not on curved surfaces, and not easily on flat angled surfaces.

closed 2026-09-29 by prusaslicer-bot[bot]