prusa3d/PrusaSlicer #9442: vertical dimension accuracy - adaptive (constant) layer height

denis-itskovich, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
When the user selects layer height - vertical dimension accuracy is not guaranteed: if the model height is not multiple of the selected layer height - the actual height might inhertently differ (mean then even when printed on the ideal printer) from the model height.

> **NOTE:** This feature is totally different from "adaptive layer height" option in variable height params: the requested feature is about keeping constant layer height (excepting may be the first layer).

**Describe the solution you'd like**
I suggest adding in slicing params, near the layer height input box, checkbox "adaptive layer height". In case when checked - the user-selected layer height will be used only as a hint, but the actual layer height will be calculated based on the provided hint and the model height.

> **NOTE:** The calculation should also take into account first layer height. The first layer height should be used precisly as defined by user, but the actual layer height for other layers will be calculated as mentioned above.

**Describe how it would work**

#### Given: 

* user-defined `layer_height`
* user-defined `first_layer_height`
* model-based `model_height`

#### Calculation:

```c
double get_layer_height(double layer_height, double first_layer_height, double model_height) {
    layer_number = (int)(model_height - first_layer_height) / layer_height);
    double actual_layer_height_max = model_height / layer_number;         // actual will be greater or equal to requested layer height
    double actual_layer_height_min = model_height / (layer_number + 1);   // actual will be less or equal to requested layer height

    // between 2 options of layer height we want to choose the closest to the user-requested one:
    double err_min = abs(layer_height - actual_layer_height_max);  
    double err_max = abs(layer_height - actual_layer_height_min);

    if (err_max < err_min) {
        return actual_layer_height_max;
    }

    return actual_layer_height_min;
}
```

Tupson444, 2023-01-29:
This is exactly what I thought a while ago. When you have an object that is 20mm tall and use 0.3 layer height, the top layer will be printed at 20.1mm, because 20 isn't divisible by 0.3. This usually doesn't matter but it's important when parts need to fit something.

You can manually change general layer height to fit the model height, but it would be much better if it was done automatically. I think this should definitely be implemented (and, while not good to be the default setting, there should be a reminder/tip about it, because if you don't think about layer height you would expect the printed height to match 3D model, until you find out the hard way).

Ioann44, 2025-05-11:
Any updates, @rtyr?

Tupson444, 2026-09-14:
still relevant