prusa3d/PrusaSlicer #9444: Funny Arachne Issue (Layer width >5mm)

TjardF, 2023-01-29:
### Description of the bug

I recently drawed the first Idea for an Adapter in Cad and wanted to have a look how the Slicer does all the tiny Details, sadly it ended in a mess...
The Linewidths are completely off...
[image]

### Project file & How to reproduce

[SLS v1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10530397/SLS.v1.3mf.zip)
Simply hit slice

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+MacOS-x64

### Operating system

Mac OS 10.15

### Printer model

Voron Trident Stealthburner

nulaft, 2023-01-29:
Your bridge flow ratio value is way too large. It is set to 95 whereas it should be between 0 and 2. You can set it to 0.95 to get a good result.

TjardF, 2023-01-29:
oh okay. was a super slicer profile, looks like import went wrong...

rtyr, 2023-01-29:
Can we close the issue?

TjardF, 2023-04-13:
sorry for late reply.
Yes we can.

closed 2023-04-13 by TjardF