prusa3d/PrusaSlicer #9448: Two bottom layers on supports

guido66611x, 2023-01-30:
Often, the single bottom layer on supports will break apart when removing the model from the bed, leaving tiny chips adhered to the bed that are extremely difficult to remove (e.g. even with a plastic razor blade).

Ideally, I would like the option to have 2+ bottom layers on support areas, so that the supports don't chip apart, and/or they are easier to remove from the bed.

Thank you!

yanfoo, 2023-02-05:
I would like the ability to add more layers too. My textured PEI likes to hold onto anything 1 layer thin.

guido66611x, 2023-03-08:
Just to be clear, since the title of my request got changed to only apply to organic supports, that I would like this feature with all kinds of supports.  That said, yes, organic supports are great, and I have been using them a lot lately, and it would probably be best implemented on those first.

ErockSaint, 2023-04-20:
I definitely feel that the bases are too thin on the organic supports. I have more failed prints, than successful ones with the organics. Mostly the whole base getting knocked over, seems like it wouldn't do that as much if there was a another layer or two.

seppelicous, 2023-05-01:
[image]
Same here, the thin raft is extremely sticky and a pain to remove. For that reason I switched back to normal supports

PanosPetrou, 2023-10-05:
I agree, organic supports need at least one more layer for better adhesion.

Tupson444, 2026-09-13:
still wanted