prusa3d/PrusaSlicer #9451: Non-interrupted ironing

k1-801, 2023-01-30:
**Context**
When there is a flat top surface on a model, designed to be ironed during a print, and some embossed writings on top of it designed to be printed in a different color, the "ironing" feature in prusa slicer will work it's way around the emossed text, even though it might be very narrow, despite it having a proper solid infill below it, ultimately increasing the print time on unwanted movements, accelerations and decelerations. 

**Is your feature request related to a problem? Please describe.**
Sometimes, then there is a portion of the area that gets ironed where the distance of one ironing pass gets really short, this area gets overheated by the nozzle passing over the same spot several times over just a few seconds. When there is an Y-shaped top surface on a part, and the ironing starts on the area of the horns, the printer will first iron one of them, then move to the second one, and only then merge the pass with the initial horn, often leaving a nasty looking ridge of plastic that was shaved off the top surface, which could be avoided if the "horns" were ironed in the same pass.

**Describe the solution you'd like**
Instead of breaking the top surface into separate areas that will be ironed independently, I suggest allowing the slicer to consider them one larger area, if the travel distance between them is short enough to iron both of them in one ironing path, thus reducing the numbers of accelerations and decelerations. If there is a solid infill in there, it should be considered a place that should be ironed, even though it's not a top surface. If there is none, the print head should continue moving without extruding (not sure if it should go at the same speed or at least try to reach regular travel speed).

**Describe how it would work**
As described above

**Describe alternatives you've considered**
An alternative here is generating a body (cube) in the slicer in the size of the layer to be ironed and setting it to "iron all solid surfaces", effectively making it iron only one top layer, but without making any breaks where there is any embossed lines. It is not as easy as just selecting the right option in the ironing section, but it does work. For the Y-shaped areas, there is no workaround.

kubispe1, 2023-01-31:
Hi, please can you attache 3mf. We will inspect your issue according to your description and your print settings. Thanks

k1-801, 2023-01-31:
[Вытяжка-фикс-вертикально.zip](https://github.com/prusa3d/PrusaSlicer/files/10545782/-.-.zip)

Here's the current project file, currently with the described workaround modifier cube in place, as well as with a workaround cube for #9450 .
Compare the resulting ironing pattern with what happens when both modifiers are removed and ironing is set to "all top surfaces" - the ironing will avoid the embossed areas at the cost of surface quality:
[image]

[Вытяжка-фикс-bad-iron.zip](https://github.com/prusa3d/PrusaSlicer/files/10545817/-.-bad-iron.zip)

The area around the oval cutouts can be made better if the whole part is rotated 90 degrees, like so:

[Вытяжка-фикс.zip](https://github.com/prusa3d/PrusaSlicer/files/10545737/-.zip)

However, it does not fix the joining ridge. Even worse, my printer has a bad design of part cooling (only blows in one direction), which is heavily required by the bridges, so the part has to be aligned with cooling direction in mind.

k1-801, 2023-02-07:
I found a model that could be a good example of a Y-shaped surface mentioned above.

[image]
[RIGHT_V2.zip](https://github.com/prusa3d/PrusaSlicer/files/10678986/RIGHT_V2.zip)

As can be seen on this screenshot, one half is ironed first, then it stops at the joint and moves to iron the second half, and only when it reaches the joint again it will merge the surfaces, which in some cases forms a vertical wall on the line where the surfaces are merged, and even if it doesn't, it wastes more time than it could.

If we had some kind of a threshold for non-interrupted ironing, set to, say, 5 millimeters, the cut in the middle of this surface would be considered a part of this top surface, and the print head could just jump it on ironing passes, yet the narrow pieces at the very left of the screenshot would still be ironed separately, as it makes no sense to try and do them in one pass. The undesired wall formation would be minimized to under 1mm of length, or would be gone completely, as there is not that much plastic there that could overheat and deform during ironing. With the threshold set to 0, it would effectively return the feature to current behavior.

Unfortunately, the algorithm for that might get really complicated, but on first glance, it seems to be worth it.

k1-801, 2023-03-04:
I suggest to close this in favor of #5409, seems to be close enough.

closed 2023-03-04 by k1-801