prusa3d/PrusaSlicer #9452: Bridge angle is not realtive

SphaeroX, 2023-01-30:
### Description of the bug

If you determine the bridge angle, this works quite well unless you rotate the object. The bridge angle then does not rotate. You can see this most clearly in the screenshot here I have marked how much a red.

[image]

### Project file & How to reproduce

[bridge.zip](https://github.com/prusa3d/PrusaSlicer/files/10538964/bridge.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Win 11

### Printer model

custom

neophyl, 2023-01-30:
None of the infill angles are linked to the objects, they are linked to global values.

kubispe1, 2023-01-30:
@neophyl is right. 
When angle = 0° - angle is automatically calculated
When the user input value it is linked to global coord. system
I am closing the issue. I think there is no bug...Right ?

closed 2023-01-30 by kubispe1