prusa3d/PrusaSlicer #9453: Feature Request - Align parts by face

thew1se, 2023-01-30:
Currently when multiple parts are placed on the build plate, or if we'd like to merge certain parts, there is no simplified way to align them as desired or to have a measured distance between them.

**Describe the solution you'd like**
To align parts, there should be a method of selecting a face of a part, then another face of another part, and then a button to "align planes".   This will add a constraint to the model when future part moves are made.   In future, aligning by vertices or edges would also be beneficial.    All 3D modelling programs have this feature for mating multiple parts.

**Describe how it would work**
Click a button to "Align Parts".  Click face 1 of part 1,  click face 2 of part 2.  Part aligns.  Immediately have a pop up which says "reverse alignment?", if yes selected, then the part of face 2 is flipped so the backside of the face is aligned with part 1.  Have a new pop up which says "Offset faces?" if yes selected then enter a measurement to offset the faces by a distance.  This will add a gap (or overlap) between the parts. If a part cannot be aligned due to previous alignments, have a pop up which says, "over constrained, remove previous alignments first"

neophyl, 2023-01-30:
As 'all 3d modelling' programs have this already then its obviously better left to cad.  After all the slicer is not a CAD package.

pointbrink, 2023-01-30:
@mix579 @neophyl PrusaSlicer has added so much functionality that could have been left for CAD programming, the program is consistently overachieving as a "slicer" into nearly a one stop shop of modifications. I don't agree with dismissing feature requests because another program is already designed for it.

jokohoko, 2023-01-30:
To reduce the request but do keep the functionality:
Add a align face to front function?
I do sometimes miss this to make sure the front face is aligned to the X axis. Z suite has this button for all directions. But just having a face front option in right click could be enough.
(align both models face front, and then rotate both to the desired alignment of each other)

mix579, 2023-01-30:
> is consistently overachieving as a "slicer" into nearly a one stop shop of modifications

Yes. And I wish they'd focus on getting the bugs out and adding core functionality. Can you say straight line seam or support painting (without having to resort to undocumented tricks)? Paint on brims? Or adding arc overhangs for something I can't do in CAD. 

Now do I feel offended when they add those features? Of course not, and I can see how for some people they may be of use. Glad to see that PrusaSlicer is under constant development.  And of course, they're adding quite interesting new features such as lightning infill (which has exceeded my expectations) and (hopefully very soon) organic supports. My point is simply that there's some basic stuff that I wish they'd pay more attention to than some of the new things they're touting, and that features readily available in CAD software should be lower on the list.

thew1se, 2023-01-30:
@mix579 @neophyl although I do agree with you both in general, it is not always a simple affair to convert a STL to a cad program (which was my first attempt prior to using prusa slicer) as an intermediary. 

That being said, I believe this request to be a basic function which compliments having more than one part on a build plate and is the natural evolution of a slicing software when working with models downloaded off the internet.   

For example, features as "copy/paste",  "add part" or "add negative volume" which suffer because there no way of placing accurately.

Speaking for the 3d printing community, not everyone has the know how or inclination to redesign every part that needs a slight modification.

jokohoko, 2023-02-01:
There is also the added usage in the future with the text embossing or library model placement.
To align text correctly to the outside geometry this feature would make it easy.

ex-nerd, 2023-10-19:
> As 'all 3d modelling' programs have this already then its obviously better left to cad

I've been fighting this in Fusion360, Onshape, and MeshMixer, and have yet to find a way to actually align two STL files (as in "select 2 faces/vertices and move the meshes so they touch and can be merged).

Use case: I have a bunch of nice files from Loot Studios that were pre-cut for smaller printers but would save a lot of time/effort/filament if I could just combine a bunch of the smaller parts into a single one and print it on my XL.

rob-miller, 2024-10-26:
I too have this exact problem with a multipart model designed to be glued together. I too went searching for a tool to help with this and gave up. 

It is reasonable to have in the slicer because it is a manipulation of an STL or similar rendered object, and the slicer does that all over. CAD programs generate STLs, and only manipulate imported ones as an afterthought, and them only with specific translations and rotations which are difficult to work out.   Only Meshlab actually manipulates STLs, but it isn't supported and doesn't do this task well (too be fair I didn't try because I didn't want to install it again).

Pick 3 vertices on each object and merge to align them, that's all.