prusa3d/PrusaSlicer #9454: Modifier presets with print settings

jokohoko, 2023-01-30:
**Describe the solution you'd like**
In the top right you now have the printer, settings and filament, with supports and brim. There i would propose to add a dropdown with checkmarks list of modifiers to apply to the settings.
For example, i often change the support settings to snug, 45° with smaller contact distance.
Or I change a range of speed settings to limit the top speed and thus noise on my mk2.5.
Or I make models hollow by changing infill, top and bottom layers and such.
Or for models with bad adhesion, apply a series of alterations to improve first layer compression.

Those modifications i do not want to do to all standard profiles. I just want to apply them over the presets. Thus you could make a bundle of mods, and apply them to any preset active independent of material or printer.

**Describe how it would work**
Not sure if a multiselect drop down is possible. But like when you right click a part to change settings to that specific part, i would see a menu where you can make a list of changes for that mod. And in the front page in that mods list you can activate a specific mod or even multiple at the same time.

[image]

**Describe alternatives you've considered**
I know this might be a big implementation on interfaces. But i imagine this being like the filament overwrite feature whereby a specific filament type can also change printer speed and retract settings if needed.

kubispe1, 2023-01-30:
Hi, thanks for the feature request. There is a duplicate issue already. #8952
I am closing it.

closed 2023-01-30 by kubispe1

jokohoko, 2023-01-30:
I feel like my proposal goes a lot further by allowing a set of mods in one selection. But the core wish is comparable indeed.

On the issue #8952 you could also propose a shortlist page of the settings you use most on one short page.