prusa3d/PrusaSlicer #9455: Adding multiple perimeters and alternating perimeters for the bottom layer in spiral vase mode

clfaye, 2023-01-31:
Prusa Slicer restricts the number of vertical shell perimeters to 1 when you are printing in spiral vase mode.  This, of course, makes sense because vase mode is one continuous line from the bottom to the top of the print and can only be one perimeter thick.  However - a vase often will have a solid bottom of a few or several layers.  It would be very useful to permit more than one perimeter on the bottom, and to be able to alternate an extra perimeter to lock in the bottom fill pattern.  This would create a much higher probability of a water tight bottom for the vase.  Cura does this very simply and therefore I use it exclusively for making vases right now..

Sergej-Popov, 2023-04-06:
I was surprised the modifier didn't allow me to do it. 

[image]

timur-tabi, 2024-08-17:
What's ironic is that there are models on printables.com that require more than one perimeter in vase mode in order to print properly, and other slicers can handle it, but not PrusaSlicer.

https://www.printables.com/model/282130-gerometric-origami-flower-pot-and-planter-for-plan

So this is definitely a feature that can be added.

timur-tabi, 2026-09-03:
This is still a useful feature to have.