prusa3d/PrusaSlicer #9456: Part autoscale forced regardless of intent.

Aquilux, 2023-01-31:
### Description of the bug

The part re-scale dialog decides for you that you want your part re-scaled.

Expected: As the phrasing of the window header implies, prusaslicer can not know weather it should scale down a part. Part re-scale dialog should have y/n selection.

Actual:
[image]

Not only is this automatic behavior annoying on it's own, it adds insult to injury by forcing you to acquiesce to it. Putting only an "ok" button on the dialog and forcing the user to click on it forces them to be an active participant in having something done to them, decided by the programmer, that they do not want. It'd be better psychologically to not have the blocking dialog at all. Either remove it, have it be disable-able in settings or add a meaningful mechanism for the user to decide what happens.

### Project file & How to reproduce

[requirement.zip](https://github.com/prusa3d/PrusaSlicer/files/10542470/requirement.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+win64-202209060714

### Operating system

Windows 10 (build 19042), 64-bit edition

### Printer model

i3 MK3S+

sarusani, 2023-01-31:
This is an information that your part has been altered (in scale) during loading. I don't think it would be a good idea to automatically alter a part without informing the user about it!

You say it's "regardless of intent". I'd assume your intent is to print a part (otherwise why load it into a slicer....). If it's to big you can't print it. So it is scaled down. What purpose would selecting "no" have?

If you want to make it bigger again. You can always set your desired scale after the part has loaded.

neophyl, 2023-01-31:
Except that you can load in a part too large for your bed and then cut it up to fit.  So this scaling it automatically on loading is not desirable at all.  You as the user should have control of the scale at all times.

We shall have to see if the restriction is still in 2.6 when it is released, especially with the new cut tools in 2.6 that have been previewed as this will make them less useful if you cant load in bigger parts to use them on.

sarusani, 2023-01-31:
You don't need a workaround or "hack". Just type the number 100 into the scale field after the part has loaded. Takes a few seconds.

And it's only scaled down if it's multiple times bigger than the print areas. Not just if it a little to big to print.

neophyl, 2023-01-31:
Quite common when printing a 6ft long spear for a cosplay lol

kubispe1, 2023-01-31:
Just to mention. This limitation is there same in 2.6.0 a1. No changes at the moment.

Aquilux, 2023-01-31:
> I don't think it would be a good idea to automatically alter a part <s>without informing the user about it!</s>

I will not accept the trend in software engineering to take control away from the user. Offer actions to the user, yes. But _**NEVER**_ decide _for_ me what I want.

> What purpose would selecting "no" have?

Just because you lack the imagination to think of a reason doesn't give you the right to dismiss what the user might want in favor of what you have decided. In my specific case I have a large assembly of 43 parts, all of which are within printable dimensions, that get exported/imported as one file then split to objects to be oriented, then arranged into separate prints (which the slicer already does automatically on auto-arrange).

>  Just type the number 100 into the scale field after the part has loaded. Takes a few seconds.

A few seconds that are entirely unnecessary if you had the option to not have the part auto-scaled. And the thing that causes the most frustration isn't that it happens, it's that from the perspective of the user the software forces them to participate in an action they do not want to perform in the first place.

Honestly this pretentious "I know better than you, you couldn't ever want to do it that way."way of thinking needs to be excised from the software dev. community. It's one of the biggest barriers to people adopting things outside of the commercial market, and what's been degrading the commercial experience as well.

Aerospacesmith, 2023-11-12:
Commenting to report that this BUG is still present in PS 2.7 beta-1. This BUG needs fixed to ask the user if they want to scale.

nyomen, 2025-02-27:
> This is an information that your part has been altered (in scale) during loading. I don't think it would be a good idea to automatically alter a part without informing the user about it!
>
> You say [...]

Can you please start understanding that someone wrote a bugticket for a reason, and then not explain him why he has no reason to create a bug ticket?!

This thing is not an information, its a change in your 3d modelfiles and you should have the possibility to say no to this, or even deactivate this functionality, i have from time to time reference shapes bigger than the model in my 3mf files. Its actualy measured sizes i put in there to use any kind of operations, or just to preserve data.
I don't know who thought of this brilliant feature, but its not needed at all. You will figure out that you cant print the object on the bed, when you try to slice it, so whats the purpse of this pre pre prevalidation that is actualy changing the saved data, most likely to something noone needs.

Take this feature out. Its actualy a bug.

nyomen, 2025-02-27:
[image]
Thats the change that needs to be done here:
https://github.com/prusa3d/PrusaSlicer/blob/e1734793e06e0fc53fc43107b2ac949aa6b99bb4/src/slic3r/GUI/Plater.cpp#L1688C1-L1691C14

For all win users: You are lucky i went through all the 3 hours building stuff and created a version for you, just replace it with your prusa slicer version:
https://limewire.com/d/d4bb4679-415d-4f66-912c-4826a0a05d00#vk5StEr3T33U1aeFlss-1FhhjYzsTDS1LuTUGpoj3uw

Elfe, 2025-03-02:
I also get the bug in 2.9.0
- loaded a small model
- scaled it up to 10150%
- trying to cut away parts

autoscale happens :(

better warn the user that a certain operation might cause issues and offer options like: cancel (cut/load), perform cut (hint on saving the project), perform cut and scale to bed

Tupson444, 2025-03-02:
I don't remember ever importing a model larger than build volume, so I haven't noticed this flaw, but I agree that it is a bug (or a wrong design choice). Even if nobody provided particular examples where scaling is not wanted at all, it should be obvious that parts should not be automatically scaled because it would be better for the user to first see how large it is, then decide what to do: a) not print it, b) scale it (deciding how much), c) switch to a larger printer (if available), d) cut it to multiple parts, or whatever else.

>You say it's "regardless of intent". I'd assume your intent is to print a part (otherwise why load it into a slicer....). If it's to big you can't print it. So it is scaled down. What purpose would selecting "no" have?

Don't assume too much (I may load a file just to see _if_ I can print it), or at least don't conclude too far based on the initial assumption (there is not only one way to solve the problem of model being too large).

Aquilux, 2026-09-24:
still a bug, still quite upset about the program deciding for me that I shouldn't want it to be scaled to the size I choose. ESPECIALLY considering there are functions specifically added to segment large models into printable pieces.