prusa3d/PrusaSlicer #9457: Wipe wall/skirt

UBTChristian, 2023-01-31:
### Description of the bug

New Prusa Slicer 2.5:
When using dual-head printing and selecting a raft, then always a wipe wall/skirt is printed around the raft.
Even the skirt/brim is set to "0".

### Project file & How to reproduce

When using dual-head printing and selecting a raft, then always a wipe wall/skirt (see attached picture, green wall) is printed around the raft.
Even the skirt/brim is set to "0".
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Windows 10

### Printer model

every dual head printer

UBTChristian, 2023-01-31:
Maybe it is a bug of the new 2.5 version.

kubispe1, 2023-01-31:
Could be Draft Shield enabled (see Printer settings/skirt and brim/Draft shield. 
In case please attach 3mf. We can check it.

UBTChristian, 2023-01-31:
No, Draft shield is disabled.

UBTChristian, 2023-01-31:
Now, I recognized that the bug is only in the German version. The English one is working fine.
I will use the English one.
Sorry and thanks.

UBTChristian, 2023-01-31:
Sorry again, now I found the problem.
The wipe wall comes along when activating the Ooze prevention and a Temperature variation of 100°C.

bubnikv, 2023-02-24:
@UBTChristian Does it mean that your problem is solved? Would you please close the issue then?

closed 2026-09-29 by prusaslicer-bot[bot]